OpenHD/OpenHD #3: Project Migration

pilotnbr1, 2019-01-17:
With a new image release the code here should reflect the latest release. How does everyone want to do this migration?

1- Leave the underlying structure the same and just bring over the relevant builder that points to @RespawnDespair repo or @user1321 repo?
pros/cons less work but we are still dancing around through multiple repos...

2- Change the builder structure so that all source (other than pi, drivers, etc) is contained in this repo?
pros/cons more work but more easily managed and workable in the long run- this raises the question as far as how to structure it- Just directories for what is currently other repos?

Other decisions will need to be made as far as promoting development here. I propose at least a master branch that reflects the latest stable release and at least one development branch. Ultimately it would be nice to see everything come under one roof. Thoughts?

pilotnbr1, 2019-01-17:
I do not mind doing as much work as I am capable of as far as this migration goes, if it turns into option 2 ( which I favor).

Anyone see other options out there?

careyer, 2019-01-17:
I also favour option 2

RespawnDespair, 2019-01-17:
I don't think it is smart to combine the builder with the other sources. More for technical reasons than anything else. Merging all the separate repositories in one central repository would be fine, but keep the builder as a separate repository.

pilotnbr1, 2019-01-17:
@RespawnDespair as the expert and creator of the builder I think your opinion carries alot of weight here.. I am totally good with that too, I just want some consolidation if its possible.

Yes21, 2019-01-17:
I'm also for having all repos (except the builder) as directories in "github.com/HD-Fpv/Open.HD".
The builder can remane on @RespawnDespair github, or be cloned on a different repo on "github.com/HD-Fpv".

I also propose to rename all directories replacing "wifibroadcast" by "Open.HD".
The builder will have to be modified slightly to be adapted to the new source structure. But I don't think it's a too hard work.

If all agree, I can create a test structure on "github.com/HD-Fpv/Open.HD" with current stable code.

But we will have to wait all WIP PRs before the definitive migration ...

Yes21, 2019-01-17:
@ALL You can now see the test I made to upload all @RespawnDespair repos to this github, with new directories names.

What are you thinking of that ?

pilotnbr1, 2019-01-17:
@Yes21 I think this is the right idea and I like it. I am curious what @RespawnDespair thinks of this. Also don’t we want all the features of @user1321 repos?

Yes21, 2019-01-18:
> @Yes21 I think this is the right idea and I like it. I am curious what @RespawnDespair thinks of this. Also don’t we want all the features of @user1321 repos?

Yes, of course we need all the features of @user1321. But I prefer to wait he has committed them to @RespawnDespair github, before cloning them in this repo.

And we also need your latest OSD improvements ...

htcohio, 2019-01-19:
@All, the

This is the repository for the "Super Image". 

*Dont be hesitant because of "ath9k" in the title, that simply means that it is the sand with changing capable build.

So, if everyone ends up agreeing, Vitaliy will need to submit a "pull requested" to Jelle, then merge with his repository along with Lukes awesome osd.

Then... a couple of us build the image tomorrow hopefully, confirm results are the same as mine today.  

Then merge Jelles repo here? 

**I'm just making sure I understand the process correctly... :) 

***I would strongly suggest that we start to release what we have when we have it instead of trying to perfect instructions and the wiki etc... I think that will all fall into place with time. 

Also, it would be really great if somebody familiar with other branches of ezwfb is able to do some testing... 

•I'm relatively confident that we can make some kind of small announcement this weekend if any of you agree.  

wherever we announce, I'm hoping that for the moment at least we can start a new thread on a different forum because rcgroups is experiencing all kinds of server problems for the last couple of months.  

I fear it could get difficult and frustrating for us and other users trying to communicate on RC groups.  I realize that it would be a change but it would also potentially expose the project to a whole new group of people :)))  

 https://github.com/user1321/wifibroadcast-image-builder/tree/user1321-5MHzAth9k?files=1

closed 2019-01-23 by pilotnbr1