OpenHD/OpenHD #4: First Image release

pilotnbr1, 2019-01-17:
We have been talking about a new image and have some really awesome features to share.

I think we can all agree that the more new features we add the more experimental the image becomes.

Do we want to release several images with different degrees of risk?

In my mind we have about 3 possible images floating around
1. most stable image just supporting new pi hardware and drivers for new wifi adapter hardware
2. beta image all of 1 plus:  with app settings, sound, flir, gpio, osd, improvements and other stuff I am forgetting that is fairly safe and stable
3. alpha image all of 1 and 2 plus: bandwidth switching 

I am ok with releasing some risky stuff (with lots of disclaimers) as I think the WBC crowd is pretty experimental oriented.. Thoughts?

htcohio, 2019-01-19:
hey, sorry I didn't notice the conversation going on here.

I will have a very good idea of everything and the whole picture after tonight into tomorrow morning.

I'm feeling that we should just release the most feature filled version that was shared on Telegram messenger.  

many of the features or more advanced enhancements do not have any effect on the performance unless someone chooses to experiment with it..

additionally, I discovered that the traditional RC that required CTS and somehow was working with rtl8812au cards did in fact begin to have glitchy trouble with a little interference and at the edge recception.

The svpcom encrypted RC seemed to work nearly as good as the original with atheros.

**I also have not tested encrypted RC with the lower 2 Mbit configured bitrate... (the costume 2/5.5/11/12 Mbit bitrate for the uplink could provide some benefit.  ( we will see).

closed 2019-02-08 by pilotnbr1

kpetrykin, 2019-04-03:
Hello! I am interested in testing Flir One functionality, but the link to download an image is broken. (https://github.com/HD-Fpv/Open.HD/wiki/Other-Projects-and-Images)
May you fix it?
Thanks!

pilotnbr1, 2019-04-03:
@kpetrykin an image should be shared soon (in the next week I hope) I think it will be linked on the wiki landing page

kpetrykin, 2019-04-03:
@pilotnbr1, i've found two folders (~/flir and ~/flirone-v4l2) in the home folder of the last Open.HD image - does it mean that some functionality is already presented in it?

careyer, 2019-04-03:
Hey @kpetrykin, thanks for your interrest in the FLIR support. The prototype for FLIR support was based upon a modified version of the lagacy EZ-WBC 1.6. We are currently working on support for FLIR within Open.HD which has a way cleaner and hopefully much better perforance due to better structured and optimized code. As @pilotnbr1 used to say: It is in the works.... so stay tuned ;-)

myeabhishek, 2019-04-10:
Hey @careyer , which flir camera are you guys trying to interface with this system?

kpetrykin, 2019-04-15:
Hello, @careyer! 
Did you include some flir one support into this image (11.04.19): https://drive.google.com/uc?id=1yWrExMswJ26ZFm3YyVhlJzwpgQYNpCcx&export=download

careyer, 2019-04-15:
Hi,

All prerequisites for usage of the FlirOne generation 2 are now in that image:

- Flir Driver
- V4l2loopback 
- omxh264enc (gstreamer)

You just need to plug together your custom video pipeline and get it transferred via tx_rawsock.
Haven't had time yet to play with those settings. But hopefully I will have some more time soon.

Cheers!

kpetrykin, 2019-05-16:
Hello, @careyer!

Are you sure you had an omxh264enc in your image?
`pi@wbc(rw):~$ gst-inspect-1.0 omxh264enc
No such element or plugin 'omxh264enc'`

The image is "Open.HD-2019-04-24.img".

careyer, 2019-05-19:
Hi Konstantin,

Your observation is right. For some reason the image is missing the omx modules. We will try to fix this in the upcoming versions. 
Thanks for bringing this up.

Cheers! 
Best regards Thomas

kpetrykin, 2019-05-20:
So, i made a translation of 160x128 thermal image with the following pipeline from cmd:
`sudo nice -n -19 gst-launch-1.0 -v v4l2src device=/dev/video3 ! 'video/x-raw,width=160,height=128,framerate=10/1' ! omxh264enc target-bitrate=100000 control-rate=1 ! 'video/x-h264,profile=high' ! h264parse ! filesink location=/dev/stdout | sudo nice -n -19 /home/pi/wifibroadcast-base/tx_rawsock -p 0 -b 8 -r 4 -f 1152 -t 1 -d 12 -y 0 c46e1f188be5`