OpenHD/OpenHD #5: Save USB Video Missing Lib

htcohio, 2019-01-23:
The save video function over USB after flight is reported not to be working, please investigate.

pilotnbr1, 2019-02-09:
Confirmed video save feature is broken. Telemetry and log files get created and save data, but no video

pilotnbr1, 2019-02-11:
Probably also limited to 4gb files per the issue on EZ

pilotnbr1, 2019-02-17:
Fixed! Thanks to Yves and Vitaliy

closed 2019-02-17 by pilotnbr1

VU2ATX, 2019-03-22:
> Fixed! Thanks to Yves and Vitaliy

Hi, I build the image on 26th eb 2018 and tested and ound that it is yet not ixed same problem (Unable to format sdcard) comes when set VIDEO_TMP=sdcard.

pilotnbr1, 2019-03-22:
Hi @VU2ATX , thanks for testing. An issue has been open for a little while regarding this. Quite honestly not sure if this will be fixed. We have recently seen the effects of depleted memory- its ugly... Suddenly the Pi will just run out of resources and crash.

Perhaps with some better safeguards built in (sush as constant checking of available free space) and thorough testing it could be a supported feature again.

VU2ATX, 2019-03-23:
> Hi @VU2ATX , thanks for testing. An issue has been open for a little while regarding saving to internal sd card. Quite honestly not sure if this will be fixed. We have recently seen the effects of depleted memory- its ugly... Suddenly the Pi will just run out of resources and crash.
> 
> Last I saw saving to a usb stick (this issue) was fixed.
> 
> Perhaps with some better safeguards built in (sush as constant checking of available free space) and thorough testing, saving to internal sd card could be a supported feature again.
> 
> Additionally the image you built might not have had the fix for saving to usb. Many fixes are first implemented in the development branch and then later merged to master branch.

Hi Pilot#1,
It was really very sad to hear that. But this feature is working seemlessly in EZWBC1.6RC6 with only limitation that ile size was limited to 4GB and only one file was saved on to the external SD card though much larger space was available on external SD card.

I did built the image on 26th feb so should have mereged in master branch as you said was fixed on 18th feb. I would re-build afresh the new image and again update here.

Summer88, 2019-12-02:
Looking for a way to fix USB video: I copied this file to local PC:
 /wbc_tmp/videotmp.raw 

and it seems that it is fine, I can play it.

In wifibroadcast-scripts/video_save_functions.sh there is no reference to it, is this supposed to be correct?

There's a reference to /tmp/videofile but this seems to be empty.

pilotnbr1, 2019-12-02:
> Looking for a way to fix USB video: I copied this file to local PC:
> /wbc_tmp/videotmp.raw
>
> and it seems that it is fine, I can play it.
>
> In wifibroadcast-scripts/video_save_functions.sh there is no [...]

Videofile var is declared in rx_functions.sh
https://github.com/HD-Fpv/Open.HD/blob/master/wifibroadcast-scripts/rx_functions.sh

Line 121

Summer88, 2019-12-03:
Thank you, I've verified USB save is working, in my case I was using an NTFS USB.
Switching to FAT32, works fine.
Telemetry, rssi, and video are saved.