OpenHD/OpenHD #7: How to connect android app to open-hd?

roque-canales, 2019-01-24:


Yes21, 2019-01-27:
Hi Roque,

The first thing is to verify the "RemoteSettingsEnabled" parameter at the end of wifibroadcast-1.txt.
1 : the communication protocole is always on
60 or 120 or ... (seconds) : the protocole will stop after 60 or 120 seconds if the app is not connected (this is the case where you don't need to use the app, and want to reduce the cpu load on a Pi zero)
0 : the protocole is off

To connect the app, just connect your android device to the groudPi wifihotspot, and start the app.

That's all.

pilotnbr1, 2019-01-28:
For anyone reading this later. Wifibroadcast-1.txt is now openhd-settings-1.txt

htcohio, 2019-04-17:
that hate this is resolved but is also really not necessary now that smartsync has been implemented...

closed 2019-04-17 by htcohio