OpenHD/OpenHD #16: RC dropout if manual Master RC MAC assigned

htcohio, 2019-02-01:
The RC drops out due to some bug when using the following settings.

RC=Range
Master MAC= (mac of most powerful ground rx/tx card)
Slave MAC= Either 2nd MAC or "0 auto".
BandSwitcher=0 (off)
*I also confirmed the same problem exists if a 3rd card is added as well on the ground.

***Note***
Master MAC input for dedicated RC uplink DOES  works fine only if BandSwitcher=1(enabled)

pilotnbr1, 2019-02-09:
Update: Justin is working on some logic that will fix this issue...

htcohio, 2019-02-11:
This issue has been resolved in the development branch.

Now, users can choose to have a single high power card on the ground (rx) side as a dedicated RC/RX transmitter.  

The Original EZ wifibroadcast project transmit at the RC based on the lowest RSSI received on the ground, this could be problematic in the event 1 or more of the ground cards are of the low-power type and although the receive signal strength is good might lack the necessary transmit power to transmit RC to the Drone side.

closed 2019-02-11 by htcohio