OpenHD/OpenHD #19: RTL8812 Higher than normal packet loss

htcohio, 2019-02-01:
rtl8812au chipsets (Alfa awus036ach) seems to have higher than normal packet loss on 5.8ghz.

2.4 ghz is not usable due to constant packet loss and bad blocks.  could be related to the modulation or MIMO RX/TX chain config in drivers.

Also, USB3.0 should be permanently disabled as it is not necessary, a high quality micro usb can be used on the alfa card instead of bulky usb 3.0 cable.

zipray, 2019-02-18:
https://github.com/aircrack-ng/rtl8812au/blob/1ae2f322fcb597c298a81f93877af277984835e2/os_dep/linux/os_intfs.c#L585

pilotnbr1, 2019-05-22:
New information suggests that cycling the card from monitor then ap then back to monitor has resolved packet loss issue. Needs to be tested

htcohio, 2019-07-12:
Closing, refer to new combined rtl8812au issue
https://github.com/HD-Fpv/Open.HD/issues/115

closed 2019-07-12 by htcohio