OpenHD/OpenHD #22: 8812AU 8814AU support MCS 9-10-11-12-13-14-15 and up

roque-canales, 2019-02-05:
Hello,

88XXAU chips can be used in MIMO mode for long range (MCS 9-10-11-12) or (VHT 2 Streams)

these board ar 2T2R.

For the moment, when using these boards only one antenna transmit. This is because we are using only one Stream mode (datarate MCS below 9)

I see that aircrack dkms 88xxAU driver is compatible with these multistreams mode (MIMO)

So it's open-hd that dont use these driver capability.

Does anybody could check this? maybe rodizio?

pilotnbr1, 2019-02-05:
It’s been discussed. Sounds like we have a guy that will play with it. Have tried that feature?

Hoping someone more knowledgeable on the specifics will comment.

zipray, 2019-02-06:
https://github.com/svpcom/wifibroadcast/commit/706af675ccb994488ded2d94e9a375191354a553

Yes21, 2019-02-19:
> [svpcom/wifibroadcast@706af67](https://github.com/svpcom/wifibroadcast/commit/706af675ccb994488ded2d94e9a375191354a553)

What do you mean ?

Yes21, 2019-02-19:
Following the advice of @svpcom, we modified the `core/rtw_xmit.c` file from the driver :
replacing 
`u8 fixed_rate = MGN_1M, sgi = 0, bwidth = 0, ldpc = 0, stbc = **0**;`
by
`u8 fixed_rate = MGN_1M, sgi = 0, bwidth = 0, ldpc = 0, stbc = **1**;`

But it didn't resolve the problem.

Do you think we could change this parameter with the .conf files ?

svpcom, 2019-02-19:
I've made an experiment: replace second antenna with terminator (50 ohm load) and run tx on mcs8 rate. **No any packets was received!** So in case of MIMO (MCS8+) you must receive signal from **all TX** antennas (which is false in case of WFB where multiple antennas using for diversity).  So you can use only STBC and not MIMO modes.

zipray, 2019-02-19:
> svpcom/wifibroadcast@706af67
> 
> What do you mean ?
@Yes21 @svpcom 
The expert's explanation is what I mean

svpcom, 2019-02-19:
@Yes21 What problem do you want to solve? Enable STBC by default (it works by patching stbc = 1 in driver) or do this in runtume (need to rewrite code with hardcoded packet headers in https://github.com/HD-Fpv/Open.HD/blob/7c4519174f63b6de4defa6d08926c18717ec7fe9/wifibroadcast-base/tx_rawsock.c#L115 and add dynamic flags evaluation like is done in my code above)

zipray, 2019-02-19:
> @Yes21 What problem do you want to solve? Enable STBC by default (it works by patching stbc = 1 in driver) or do this in runtume (need to rewrite code with hardcoded packet headers in 
>   
>     
>       Open.HD/wifibroadcast-base/tx_rawsock.c
>     
>     
>          Line 115
>       in
>       7c45191
>     
>     
>     
>     
> 
>         
>           
>            0x04, 0x80, 0x00, 0x00, // <-- radiotap present flags (rate + tx flags) 
>         
>     
>   
> 
>  and add dynamic flags evaluation like is done in my code above)
like this?
https://github.com/svpcom/wifibroadcast/blob/212f5c1d0151ddeba21ae19596e95bee80725887/src/wifibroadcast.hpp#L77

zipray, 2019-02-19:
> @Yes21 What problem do you want to solve? Enable STBC by default (it works by patching stbc = 1 in driver) or do this in runtume (need to rewrite code with hardcoded packet headers in [...]

https://github.com/svpcom/wifibroadcast/blob/212f5c1d0151ddeba21ae19596e95bee80725887/src/wifibroadcast.hpp#L77

zipray, 2019-02-19:
> @Yes21 What problem do you want to solve? Enable STBC by default (it works by patching stbc = 1 in driver) or do this in runtume (need to rewrite code with hardcoded packet headers in 
 `https://github.com/HD-Fpv/Open.HD/blob/7c4519174f63b6de4defa6d08926c18717ec7fe9/wifibroadcast-base/tx_rawsock.c#L115` and add dynamic flags evaluation like is done in my code above)

Yes21, 2019-02-20:
> @Yes21 What problem do you want to solve? 

The problem is to make the 2 antennas of AWUS036ACH (4 for AWUS1900) sending data in monitor mode.

> Enable STBC by default (it works by patching stbc = 1 in driver) or do this in runtume (need to rewrite code with hardcoded packet headers in
> 
> [Open.HD/wifibroadcast-base/tx_rawsock.c](https://github.com/HD-Fpv/Open.HD/blob/7c4519174f63b6de4defa6d08926c18717ec7fe9/wifibroadcast-base/tx_rawsock.c#L115)
> 
> Line 115 in [7c45191](/HD-Fpv/Open.HD/commit/7c4519174f63b6de4defa6d08926c18717ec7fe9)
>  0x04, 0x80, 0x00, 0x00, // <-- radiotap present flags (rate + tx flags) 
> and add dynamic flags evaluation like is done in my code above)

Do we just need to replace :
`0x04, 0x80, 0x00, 0x00, // <-- radiotap present flags (rate + tx flags)`
by :
`0x00, 0x80, 0x08, 0x00, // <-- radiotap present flags:  RADIOTAP_TX_FLAGS + RADIOTAP_MCS`
??

svpcom, 2019-02-20:
``` .c
#define IEEE80211_RADIOTAP_MCS_HAVE_BW    0x01
#define IEEE80211_RADIOTAP_MCS_HAVE_MCS   0x02
#define IEEE80211_RADIOTAP_MCS_HAVE_GI    0x04
#define IEEE80211_RADIOTAP_MCS_HAVE_FMT   0x08

#define IEEE80211_RADIOTAP_MCS_BW_20    0
#define IEEE80211_RADIOTAP_MCS_BW_40    1
#define IEEE80211_RADIOTAP_MCS_BW_20L   2
#define IEEE80211_RADIOTAP_MCS_BW_20U   3
#define IEEE80211_RADIOTAP_MCS_SGI      0x04
#define IEEE80211_RADIOTAP_MCS_FMT_GF   0x08
#define IEEE80211_RADIOTAP_MCS_HAVE_FEC   0x10
#define IEEE80211_RADIOTAP_MCS_HAVE_STBC  0x20

#define IEEE80211_RADIOTAP_MCS_FEC_LDPC   0x10
#define	IEEE80211_RADIOTAP_MCS_STBC_MASK  0x60
#define	IEEE80211_RADIOTAP_MCS_STBC_1  1
#define	IEEE80211_RADIOTAP_MCS_STBC_2  2
#define	IEEE80211_RADIOTAP_MCS_STBC_3  3
#define	IEEE80211_RADIOTAP_MCS_STBC_SHIFT 5

#define MCS_KNOWN (IEEE80211_RADIOTAP_MCS_HAVE_MCS | IEEE80211_RADIOTAP_MCS_HAVE_BW | IEEE80211_RADIOTAP_MCS_HAVE_GI | IEEE80211_RADIOTAP_MCS_HAVE_STBC | IEEE80211_RADIOTAP_MCS_HAVE_FEC)

// For MCS#1 -- QPSK 1/2 20MHz long GI + STBC
#define MCS_FLAGS  (IEEE80211_RADIOTAP_MCS_BW_20 | (IEEE80211_RADIOTAP_MCS_STBC_1 << IEEE80211_RADIOTAP_MCS_STBC_SHIFT))
#define MCS_INDEX  1

static uint8_t radiotap_header[]  __attribute__((unused)) = {
    0x00, 0x00, // <-- radiotap version
    0x0d, 0x00, // <- radiotap header length
    0x00, 0x80, 0x08, 0x00, // <-- radiotap present flags:  RADIOTAP_TX_FLAGS + RADIOTAP_MCS
    0x08, 0x00,  // RADIOTAP_F_TX_NOACK
    MCS_KNOWN , MCS_FLAGS, MCS_INDEX // bitmap, flags, mcs_index
};
```

svpcom, 2019-02-20:
This code is for set **MCS modes** (802.11n), not for **legacy modes** (802.11a/b/g)

svpcom, 2019-02-20:
for MCS#1 -- QPSK 1/2 20MHz long GI **without STBC**
``` .c
#define MCS_FLAGS  (IEEE80211_RADIOTAP_MCS_BW_20)
```

for stbc + ldpc:
``` .c
#define MCS_FLAGS  (IEEE80211_RADIOTAP_MCS_BW_20 | (IEEE80211_RADIOTAP_MCS_STBC_1 << IEEE80211_RADIOTAP_MCS_STBC_SHIFT) | IEEE80211_RADIOTAP_MCS_FEC_LDPC)
```

roque-canales, 2019-02-21:
Hello SVP, so with this mod,, 88XXau board transmit same radio data through there 2 or 4 antennas ?

Idea is to have capability to receive data on the ground even if one antenna is NLOS.....

svpcom, 2019-02-21:
@roque-canales Yes, see https://en.wikipedia.org/wiki/Space%E2%80%93time_block_code for reference.

roque-canales, 2019-02-21:
Thank you,

Yes is exactly this feature that I wait to be released on image.

I'm not expert user for compile image with this feature.

Could you please release an image with this feature activated for 88xxAU users ?

Yes21, 2019-02-21:
I'm trying to implement this code in Open.HD.
Please wait. I will share an image if it works ...

Yes21, 2019-02-22:
@roque-canales, if you want to test it, no need to build an image.
Just replace the "/home/pi/wifibroadcast-base/tx_rawsock.c" file by the one attached here, and execute "sudo make" in the "wifibroadcast-base/" directory. This should be done from a running Pi.

[tx_rawsock.c.txt](https://github.com/HD-Fpv/Open.HD/files/2895239/tx_rawsock.c.txt)

(delete the .txt extension)

svpcom, 2019-02-22:
This fragment is invalid:
``` .c
static u8 u8aRadiotapHeader[] = {
	 0x00, 0x00, // <-- radiotap version
    0x0d, 0x00, // <- radiotap header length
    0x00, 0x80, 0x08, 0x00, // <-- radiotap present flags:  RADIOTAP_TX_FLAGS + RADIOTAP_MCS
    0x08, 0x00,  // RADIOTAP_F_TX_NOACK
    MCS_KNOWN , 0x00, 0x00 // bitmap, flags, mcs_index
};
```
If you set MCS_KNOWN you **must** use MCS modes, but not legacy rates:
```
{
    0x00, 0x00, // <-- radiotap version
    0x0d, 0x00, // <- radiotap header length
    0x00, 0x80, 0x08, 0x00, // <-- radiotap present flags:  RADIOTAP_TX_FLAGS + RADIOTAP_MCS
    0x08, 0x00,  // RADIOTAP_F_TX_NOACK
    MCS_KNOWN , MCS_FLAGS, MCS_INDEX // bitmap, flags, mcs_index
};
```

Yes21, 2019-02-22:
> If you set MCS_KNOWN you **must** use MCS modes, but not legacy rates:
> 
> ```c
> {
>     0x00, 0x00, // <-- radiotap version
>     0x0d, 0x00, // <- radiotap header length
>     0x00, 0x80, 0x08, 0x00, // <-- radiotap present flags:  RADIOTAP_TX_FLAGS + RADIOTAP_MCS
>     0x08, 0x00,  // RADIOTAP_F_TX_NOACK
>     MCS_KNOWN , MCS_FLAGS, MCS_INDEX // bitmap, flags, mcs_index
> };
> ```

Ok, with this settings, it seems that the 2 antennas of 036ACH are sending !
A great step in the right direction ...

> 
> Also you need to rewrite `packet_header_init(uint8_t *packet_header, int type, int rate, int port)` to set rate via MCS_FLAGS and MCS_INDEX fields but not with legacy rates.

I will look for this later ...

And many thanks Vasily for your help and your patience !!

roque-canales, 2019-02-25:
Hello,
@svpcom:
Also you need to rewrite packet_header_init(uint8_t *packet_header, int type, int rate, int port) to set rate via MCS_FLAGS and MCS_INDEX fields but not with legacy rates.

could you please tell me the values that need to be replaced ?

Kind regards

svpcom, 2019-02-25:
replace first switch with 
``` .c
u8aRadiotapHeader[MCS_FLAGS_OFF] = mcs_lags;
u8aRadiotapHeader[MCS_IDX_OFF] = mcs_index;
```

roque-canales, 2019-02-25:
Thank you :)  ok and after?

svpcom, 2019-02-25:
You need to replace
``` .c
switch (rate) {
...
}
```
which this two lines.
mcs_flags is desired flags (depend what you want to enable: stbc, lpdc, 20 or 40 mhz channels)
mcs_index is number of mcs mode (from 0 to 7).

roque-canales, 2019-02-25:
Ok so I supress the first switch, so the switch rate by the two lines you told upper.

And after ? the code is ok or it need another upgrade?

I want to use stbc 20mhz.

roque-canales, 2019-02-25:
@Yes21 
It test your file but I Don't see any change on output.

Wich datarate you set? I test with 4 with no success.

zipray, 2019-02-25:
> @Yes21
> It test your file but I Don't see any change on output.
> Wich datarate you set? I test with 4 with no success.
Need delete tx rawsock.o and execute "sudo make tx_rawsock" in the "wifibroadcast-base/" directory.

Yes21, 2019-02-25:
> @Yes21
> It test your file but I Don't see any change on output.
> 
> Wich datarate you set? I test with 4 with no success.

There was an error in it.
Here is the later version :
[tx_rawsock.c.txt](https://github.com/HD-Fpv/Open.HD/files/2902088/tx_rawsock.c.txt)

roque-canales, 2019-02-26:
@svpcom 

Hello Vasily, thank you so much for yesterday, now I'm trying to finish de rawsock file.

Could you please help me for activate STBC 20mhz.?

I bypass switch rate:
/* switch (rate) {……………….}*/

I write theses two lines below:

u8aRadiotapHeader[MCS_FLAGS_OFF] = mcs_flags;
u8aRadiotapHeader[MCS_IDX_OFF] = mcs_index;

and after what I must to do?

Thank you so much for help us.

zipray, 2019-03-01:
> @Yes21
> It test your file but I Don't see any change on output.
> Wich datarate you set? I test with 4 with no success.
> 
> There was an error in it.
> Here is the later version :
> tx_rawsock.c.txt

https://github.com/Toradex-Apalis-TK1-AndroidTV/rtl8812_au_tx_power/blob/15e55fc8ab3940007c16b7cf391a129012582245/include/ieee80211.h

roque-canales, 2019-03-02:
@svpcom

Hello Vasily could you please tell me the good values for activate stbc with 20mhz channel and datarate 4 equivalent rate?

Yes21, 2019-03-02:
> https://github.com/Toradex-Apalis-TK1-AndroidTV/rtl8812_au_tx_power/blob/15e55fc8ab3940007c16b7cf391a129012582245/include/ieee80211.h

Hi @zipray, what do you want to show us with this link ?

svpcom, 2019-03-03:
@roque-canales use 
```
#define MCS_FLAGS  (IEEE80211_RADIOTAP_MCS_BW_20 | (IEEE80211_RADIOTAP_MCS_STBC_1 << IEEE80211_RADIOTAP_MCS_STBC_SHIFT))
#define MCS_INDEX  _mcs_index_from_table_
```
where `_mcs_index_from_table_` is from http://mcsindex.com/
See orange column 20Mhz GI = 800ns

roque-canales, 2019-03-04:
Thank you Vasily

htcohio, 2019-07-12:
Closing, refer to new combined rtl8812au issue
https://github.com/HD-Fpv/Open.HD/issues/115

closed 2019-07-12 by htcohio