OpenHD/OpenHD #24: Inav compass

pilotnbr1, 2019-02-08:
With mavlink on an inav FC the compass does not update poroperly. It slightly wiggles so it is getting info but something is wrong obviously.

pilotnbr1, 2019-03-04:
Inav heading is outputting exact heading but then the script is dividing by 100. This results in the osd seeing heading of 123 as 1.23

We have some flight controller differences here that are not following mavlink standards

pilotnbr1, 2019-03-05:
The fix involved adding another setting to OSDCONFIG.txt 
#define COMPASS_INAV  // set to true to fix Inav vs Ardupilot digit difference

This fix needs to be tested by a non inav autopilot user...

htcohio, 2019-03-06:
I am reopening this because it appears to have had an effect on the regular "pixhawk" setup compass... (There are just a bunch of rapidly changing numbers under the compass dial now) I confirmed that the osd config is not set false for inav stuff...)

pilotnbr1, 2019-03-06:
Should be fixed. Instead of if looking for whether compass_inav is set it is now looking for a true or false. More verbose now. Still needs testing from non inav autopilot

closed 2019-04-07 by pilotnbr1