OpenHD/OpenHD #27: Raspi2png might be missing

pilotnbr1, 2019-02-17:
Will need to be added back in the builder

Yes21, 2019-02-17:
I was able to install raspi2png this way: 
```
sudo apt-get install libpng12-dev
git clone https://github.com/AndrewFromMelbourne/raspi2png.git
cd raspi2png
make
sudo make install
```
But the installation of libpng12-dev did remove 6 packages which are needed for OSD. After that the OSD doesn't work anymore !!

Yes21, 2019-02-18:
I could install raspi2png without installing libpng12-dev (because it was already installed by the builder !!)
Raspi2png is working, but no screenshots are recorded ...

pilotnbr1, 2019-02-18:
Well thats an improvement!

pilotnbr1, 2019-02-21:
@yes21 I created a sub branch of dev branch in the builder in order to correct this. I added your code there and we can hopefully get screenshot working again.

pilotnbr1, 2019-02-22:
THanks for the help @user1321 ! You saved me some time! Fixed now!

closed 2019-02-22 by pilotnbr1