OpenHD/OpenHD #32: Sever on Open.HD ground pi

Consti10, 2019-02-18:
**1) Overview**
I want to create a simple server program for the Open.HD pi that makes it easier for external devices to query Informations about the system, change settings and communicate over a lossless TCP connection.

**2) Implementation:**
The program should listen for incoming TCP connections on a fixed port. This will be implemented as a simple main loop like most Open.HD scripts.
For communication I am going to use a simple text -based protocol. Unless I find another library for that, I'l just send utf-8 strings over TCP, terminated by a newline character.

 _Milestone 1:_
Query the Open.HD version / build number.
The server should respond to 'VERSION \n' with 'Version X\n'
More milestones will follow shortly.

pilotnbr1, 2019-02-18:
A consideration might be to change the format of the settings txt files. Go to something a little less human readable but better for data. Especially the joystick, apconfig, and osd settings files your average user does not play with.

Hopefully the app can become the primary way to change any setting in the future.

htcohio, 2019-07-12:
This could be good for connecting to the camera for the various Raspberry Pi camera apps that are available by simply inputting the IP address of the Raspberry Pi it's connected to.

Unfortunately all of the TX rostock and raspivid functions need to be terminated in order to change the camera parameters.

Also, we must have a way to completely terminate the 2way links as maintaining a standard two-way link even if they are just bi-directional pings has a detrimental effect on packet loss.

RespawnDespair, 2020-12-04:
QOpen.HD has much of this functionality.
I'm assuming you agree this can be closed @Consti10 ?
If not, please feel free to reopen.

closed 2020-12-04 by RespawnDespair