OpenHD/OpenHD #33: Function not working "Disable Screen Text Notifications"

htcohio, 2019-02-19:
See Screenshot, This function does not seem to work, Too many "hotspot connect/disconnect" notifications at times and it interrupts the osd so this should probably be addressed eventually...

[image]

pilotnbr1, 2019-02-25:
Should be fixed.. needs testing

pilotnbr1, 2019-02-26:
To test must pull dev branch and quiet=y in settings. Then disconnect and reconnect hotspot. You should see no message.. note some errors are still allowed to write

pilotnbr1, 2019-03-03:
Mostly Fixed. I made it handle the very common thus frequent messages that are annoying. The more serious error messages are still allowed to draw over osd.

closed 2019-03-03 by pilotnbr1