OpenHD/OpenHD #34: Aerodump (Scans all channels, does not seem to work)

htcohio, 2019-02-19:
[image]

Aerodump seems to run through channel scans but does not end up displaying any useful information afterwards...  This did work well with 1.6 and I believe some people still use it. If we do not want to continue using it, we could remove it all together...?? Thoughts???

pilotnbr1, 2019-02-22:
I think this might now work as it relied in raspi2png

pilotnbr1, 2019-02-24:
Airodump works but with many WiFi networks the results scroll in an unreadable way. I think it needs an enhancement rather than bug fix

htcohio, 2019-02-25:
I will try to verify this today

RespawnDespair, 2020-12-04:
Not really relevant anymore. Feel free to reopen if anyone still has issues with this.

closed 2020-12-04 by RespawnDespair