OpenHD/OpenHD #42: Confirm Ground Pi stops sending "Pings" when Remote Settings Disabled

htcohio, 2019-02-25:
It was previously shown that GroundPi continued to send "Pings" to air for remote settings even when "RS timeouts" It would be better if the RS terminates on ground completely when timer is set for RS

pilotnbr1, 2019-02-25:
I still don’t understand the functionality of timer.. Is the timer function supposed to kill pings so therefore also kill the ability to establish a connection? 

I guess I don’t really understand the whole sequence and flow of data

htcohio, 2019-02-25:
The timer is in place for 2 reasons that relate to the remote settings
through the app.

1. In the case of rpi zero, the cpu is 5-15% higher with two way
communication remote settings enabled.  So, the countdown timer gives users
enough time to connect the app if their intention is to change settings as
such, if that is not the intention, by the time a UAV is ready to take off
the timer runs out and you save the CPU usage without having to do anyting.

2. None of the settings changed in the app can be changed and applied while
the Drone is in the air so, generally speaking it's best to kill any
unneeded data or bi-directional pings in order to maximize rxtx performance
at both ends.

If you connect wireshark and "listen" to ground side (air is powered off
and USB joystick unplugged) on 1.6rc6 no data is transmitted from the
ground.

However, on open.hd with the same configuration and, remote settings past
60sec timeout or 0-Disabled... Wireshark picks up pings (or some other
small data packets)s from the ground side  that we suspect is the remote
settings program that does not get terminated on the ground with the air
when the timer runs out...

I hope this helps explain.

user1321, 2019-02-26:
Good day.
I added to development branch code that will start "RemoteSetting" only if it enabled.

In case if it enabled with timer:
At air side it will completely turn off after certain amount of time.
At ground side it will stay enabled.. But, as main goal save CPU at air, I think we can close that issue.

pilotnbr1, 2019-02-26:
Thanks for the explanation htcohio and nice fix user1321! Closing issue

closed 2019-02-26 by pilotnbr1