OpenHD/OpenHD #43: Relay mode fails when app settings changes

pilotnbr1, 2019-02-25:
Reported by Zipray.

user1321, 2019-02-25:
What is linkpi? Thx.

pilotnbr1, 2019-02-25:
I think I read zipray use the term ! Link as in the relay pi. Or the middle pi between ground and air

pilotnbr1, 2019-02-25:
Looks like zipray submitted a pr that miggt fix it

user1321, 2019-02-25:
Don't understand how shared memory , tx_rawsocket and change setting app linked (

user1321, 2019-02-25:
Uh, it got it. Settings change app not linked to that problem directly.
If shared memory not initialized .
Absolutely same will happen if he will edit settings change file directly using PC.

pilotnbr1, 2019-03-03:
Ziprays PR is merged.. need testing. honestly not even 100 percent sure what Zipray was saying the problem was in the first place.

Hoping some testing reveals everything is working.

pilotnbr1, 2019-04-24:
Anyone have any feedback on this? Not sure if it works as intended or not...

pilotnbr1, 2019-05-24:
Closing this.....

pilotnbr1, 2019-06-16:
Reopening this as rcgroupd user reported it is not working....

pilotnbr1, 2019-07-01:
This is duplicate. other issue has more info

htcohio, 2020-02-19:
relay is a feature that is somewhat difficult to implement with the way open HD architecture currently is setup. 

This feature worked fine on the WFB 1.6 version but, has never worked on open HD.

I am reopening this issue because, it is a feature that brings a lot of benefit for long-range fpv setups on a tripod. 

There have been nearly a dozen requests from various users and, if possible it's something that we should try to reintegrate.

closed 2022-10-16 by raphaelscholle