OpenHD/OpenHD #48: App issue

Consti10, 2019-02-26:
Just found an issue in the current change-settings-scripts implementation that is worth mentioning:
(It will be fixed in the new implementation, but until then keep it in mind:)

Changing the FREQ via the app overrides the FREQSCAN value.
Example:
Before:
FREQ=2372
FREQSCAN=N

After:
FREQ=2372
FREQ=2372

Unfortunately FREQSCAN starts with FREQ and is therefore overridden by the script

pilotnbr1, 2019-02-26:
Out of curiosity does the app also overwrite relay_freq (might be spelled differently) incorrectly?

Consti10, 2019-02-27:
It could be, but I didn't check.
There is
RELAY=N
RELAY_FREQ=12345
When changing the value for RELAY via the app, RELAY_FREQ can be overridden.
But not the other way around.

pilotnbr1, 2019-04-18:
If this is fixed can this issue be closed?

user1321, 2019-04-19:
Hi!
I can`t reproduce that.
Hover I can`t see inside github history any changes inside RemoteSettings.py
Probably, we can close that.

closed 2019-05-22 by pilotnbr1