OpenHD/OpenHD #56: Pi 3b+ Errors over time

pilotnbr1, 2019-03-12:
3 cards trigger errors after approximately 30 min runtime.

careyer, 2019-03-12:
Maybe this problem is linked to the Interrupt problem when running too many USB devices? Remember the severe joystick problems I had with the GeeekPi Pi Joystick shield with 3 Atheros cards + Joystick shield? This induced massive problems and the problem got even worse when connecting a USB keyboard on top of that. When reducing the number of USB devices the problem faded away. Just a thought!

pilotnbr1, 2019-03-14:
First and foremost good debugging and reading logs.

Ideas For testing:
* 3-4 88xxau cards to a ground RPI 3 B+ and run the receiver to see if it continues running without crashing. Until now only tested with ath9k

* build system from 0 without any kernel patches and check how stabel  it will run? Especially without au8814 ( using Ath9k band will be limited to 1-11 or 13 ch. But, it's interesting to check this first. )

* build kernel 4.9.35 and check if minor packet lost will go away or not. ( 3b+ not supported ) that will help*  to understand if problem with kernel or software

* check what is the difference beetween 3 different kernel branches: 4.9.35 ( no packet loss) 4.14 ( we use it now ) and latest 4.19?

* underclocking

pilotnbr1, 2019-03-16:
As of now problem seems mostly related to recorded video eating all ram...

user1321, 2019-03-18:
I think we can close that issue.
It was related to /etc/fstab and /wbc_tmp with size of 630M
Reducing size to 256 solve problem in case if ground board have 1GB of RAM.
That may happen again in case if someone will run ground unit at RPi 1, or any other board with RAM less than 1GB.
Solution for low RAM boards can be:  to use a USB drive as a video destination.

closed 2019-03-18 by pilotnbr1