OpenHD/OpenHD #57: Latency display

pilotnbr1, 2019-03-16:
Implement a way to display latency on the GroundPi.

Better yet would be a way to detect and fix extreme latency and provide warning.

htcohio, 2019-04-17:
I believe that this issue is more so on the wireless driver side where the frames are stuck in the Wi-Fi card buffer.  it would be nice if somebody could come in with some expertise on this.

pilotnbr1, 2019-04-30:
In high latency events it has been observed that the telemetry stream is relatively unaffected. Propose sending airpi current system time as a readable var in the telemetry stream. Timestamp video frames. On GroundPi compare the telemetry time vs the frame time. If the 2 times are too far apart then warn user

raphaelscholle, 2022-10-16:
2.2.x can display all mesurable latency in a widget

closed 2022-10-16 by raphaelscholle