OpenHD/OpenHD #58: Not booting on RPi 3a+

stxShadow, 2019-03-18:
Good morning,

i checked out the sources yesterday and build a fresh Image. This Image is not booting on a RPi 3A+.
I tried the same Image (on the same SD) with a Pi Zero and a Pi3 -> and it works without problems.

Just to be sure i used the image on a second RPi 3A+ with the same results. It's not booting.

Maybe someone can have a look at this ? 

Jens

pilotnbr1, 2019-03-18:
Which branch were you building Jens?

stxShadow, 2019-03-18:
just a normal checkout of "master".

Jens

htcohio, 2019-03-18:
@stxshadow,

Thank you for bringing this up.  I can confirm that it is not working on the development branches.  I do have a video of it working on February 2nd, since then we implemented the audio as well as dual framebuffer DSi and HDMI display.

So, one of those two things probably broke the RPI 3A+...

This is a February 21st image from the RC groups.  feel free to try this out to see if it works with the A+ for the moment.

https://drive.google.com/file/d/17inRIfB4Jq4zD61Da9_SyNUeUdJLVpE1/view?usp=drivesdk

user1321, 2019-03-19:
Please, be very careful and in case if you plan to use 3A+ as a ground unit. It have half of RPi 3 RAM - 512MB instead of 1024MB It can and probably will cause a problems if you will run it as a ground unit.
In overall, I don`t recommend to use it as ground unit at this moment.

As for broken image for 3A+,
Can you please edit config.txt file at SD card and replace last 3 lines from this:

[pi3+]
start_file=1to3b_x.elf
fixup_file=1to3bup_x.dat

to that:

[pi3+]
start_file=start_x.elf
fixup_file=fixup_x.dat

That probably will break dual HDMI + DSI video output, but in that way we can confirm that problem exactly in that place.
Thank you.

htcohio, 2019-03-19:
hey,

User1321 recommended change works ;) I just tested it on air side.

I will also confirm it's working on the ground then we can probably close the issue.

users will just be recommended not to try using the duel DSi and HDMI display due to the limited memory ram. 

It is my assumption that have majority of people intend to use the A+ on the air anyways.

stxShadow, 2019-03-19:
Will test this this evening. The Pi3A+ will be a new Air Unit. I used RPi Zeros till now.

Jens

stxShadow, 2019-03-20:
The above fix works for me too ...... my Pi3 A+ is booting now. 
DSi and HDMI is not necesarry for air units :) so resolved for me 

Thx a lot !

pilotnbr1, 2019-03-20:
Have fun!

closed 2019-03-20 by pilotnbr1