OpenHD/OpenHD #69: REQ: Add support for Frsky F.PORT protocol

TheGuv, 2019-04-16:
Frsky F.PORT is a newer protocol somewhat like S.BUS and S.PORT combined into a single-wire bus using a standard serial speed of 115200, 8N1. 

The official protocol documentation is here: https://github.com/betaflight/betaflight/files/1491056/F.Port.protocol.betaFlight.V2.1.2017.11.21.pdf

and Betaflight overview here: https://github.com/betaflight/betaflight/wiki/The-FrSky-FPort-Protocol

Implementing this in Open.HD would add the possibility of Frsky users having telemetry which is currently unavailable due to the 'special' baud rate used for S.PORT. The RC portion of F.PORT is already well-handled by e.g. Inav on standard flight controllers. Telemetry would allow people to use the radio as a joystick while still having the usual artificial horizon, alarms, voice confirmation of mode changes and all the lovely things normal Frsky RC control offers.

I am willing to help test should this be implemented sometime.

htcohio, 2019-04-17:
Hey, the sounds like a good enhancement, at the moment we are trying to optimize various aspects of the open.HD system.  once we have those worked out in the near future we can come back to this.

RespawnDespair, 2020-12-04:
Re-evaluate F.Port in 2.1, attaching this issue there.

raphaelscholle, 2022-10-16:
currently we do not plan to support any other telemetry/rc protocolls then Mavlink, issue can be reopened if things change

closed 2022-10-16 by raphaelscholle