OpenHD/OpenHD #70: add STBC to 88XX boards

roque-canales, 2019-04-17:
Hello,

I test the last revision (wiki download link) and STBC is not activated….

This feature permit to use all the antennas of the 88XX chipset for best performance….

Vasily and Yes21 made it work….

Could you please add it to next official release?

zipray, 2019-05-12:
https://github.com/HD-Fpv/Open.HD/blob/7daa426e88433f61ff0e1db76e6b9d5c962cdac3/RemoteSettings/Air/TxSecondaryCamera.sh#L11
@roque-canales Now you may need to add the -s 1 parameter to enable STBC when usb camera

roque-canales, 2019-05-12:
Hello thank you, but how to enable stbc for normal picamera?

htcohio, 2019-07-12:
Closing, refer to new combined rtl8812au issue
https://github.com/HD-Fpv/Open.HD/issues/115

closed 2019-07-12 by htcohio