OpenHD/OpenHD #71: Open.HD April 14 release does not show barometric altitude on the OSD

flyingw88, 2019-04-17:
The barometric altitude does not update with either imperial or metric units.

pilotnbr1, 2019-04-18:
Looking into this I see references td->baro_altitude in telemetry.c and in render.c however mavlink.c shows no references. It is completely absent of any baro reference. Need to take a look at a EZ 1.6 image to restore this functionality as mavlink.c on the ez repo seems to be pre EZ 1.5.

pilotnbr1, 2019-04-23:
Upon more investigation looking at EZ 1.6 baro_alt headers are there but in mavlink.c there is no reference.
Was baro_alt working in EZ 1.6? 
Is baro_alt somehow being pulled from some other source?

For now I am going to tie baro_alt to vfr hud alt which I believe is baro alt on most FC. Will test and see.

If that does not work will use-
ADSB_ALTITUDE_TYPE_PRESSURE_QNH
Altitude reported from a Baro source using QNH reference

I just do not want to add a whole new message type and thus a new case

pilotnbr1, 2019-04-23:
Now seems to be working. Build Dev branch to try. Closing issue

pilotnbr1, 2019-04-25:
Once gps comes online baro shows absolute altitude rather than relative.

pilotnbr1, 2019-06-09:
Its become apparent that there been some conflation of the terms "baro", relative, msl, etc

in the end the most accurate altitude readings are the derived altitudes from the fusion of all the sensors. Probably global messgaes would be best.

I propose displaying relative altitude (mavlink global or vfr hud) as the main altitude displayed. 
Then on the smaller altitude display (where baro alt used to be) show amsl altitude

closed 2019-06-09 by pilotnbr1