OpenHD/OpenHD #75: no data  of MAVlink in Pi 3B+

HFUTLCW, 2019-04-26:
The osd is no working in Pi3b+ ,I can't get any data from mavlink.The  Proper Wiring must been right.

pilotnbr1, 2019-04-26:
Most of us use pi3b+ and mavlink. So you have that going for you!

We need more information to help you.
What flight controller?
What software on flightcontroller? (Ardupilot Inav etc)
What Airpi?
What wifi cards?
Are you seeing video?
Does the osd display over the video?

You sound sure of your wiring. Confirm hou chose "mavlink" as telemetry option in the openhd settings file. Confirm your baudrates match on the airpi groundpi and the flight controller. If using ardupilot you might have to select a stream rate for telemetry.

HFUTLCW, 2019-04-26:
Sorry for my mistakes.I fond that the Baud rate is wrong.I changed the Baud rate in the configur text.Now,it is working.Sorry again for my dump question.I am freshman in github .This project 
 is so useful.I wish this project can be 
 continued.Thank you for your patient reply.

pilotnbr1, 2019-04-26:
No problem! That was an easy one! Closing this issue

closed 2019-04-26 by pilotnbr1