OpenHD/OpenHD #76: 4/24 image unable to ssh via ethernet

htcohio, 2019-05-01:
I just realized that I cannot SSH to air or ground on the April 24th image.

pilotnbr1, 2019-05-23:
By default raspian does not enable ssh for security reasons.
To enable it you must have a file called “ssh” in boot partition- it’s there
Also enabled ssh via raspi-config

Eth0 shows that it’s up and configured . It just doesn’t have an IP address assigned. I was able to write an ip to the interface but it was not persisting beyond the session... The issue is somewhat complicated by main.sh does some manual configuring of eth0

On another note we are adding package udhcpd which is depreciation and been rolled into busybox. Will try later with busybox...

pilotnbr1, 2019-06-03:
closing this as it seems to have corrected itself...

closed 2019-06-03 by pilotnbr1