OpenHD/OpenHD #85: Upgrade RTL8812AU Drivers?

htcohio, 2019-06-02:
Im just creating an issue for the known RTL88xxau 
V4.2.20 driver we have been using for the last 6 months. 

There are 2 newer branches 5.3.4 and 5.6.4...  I believe 5.3.4 is considered to be stable, I've also followed they gitub issues and it seems like the monitor mode problems have been resolved if we can be on the correct branch...

What in general is holding us up from upgrading this? Or at least testing it out on a branch? 

*just trying to keep track of all of this and make sure it's in our sights.

Other than some basic organization on the wiki and making sure that our branches between Master development and others are organized, this has to be the most important thing we need to resolve. 

THIS WOULD BE A GREAT FIRST ISSUE FOR A NEW DEVELOPER INTERESTED THE PROJECT!!!

pilotnbr1, 2019-06-03:
Opened a new branch in the builder with support for 5.3.4 88xx .. I am also having it build drivers for 8821au.

NOTE to self: when/if we upgrade the pi to 64 bit kernel, all of the 88xx makefile patches will have to be played with again

htcohio, 2019-07-12:
Closing, refer to new combined rtl8812au issue
https://github.com/HD-Fpv/Open.HD/issues/115

closed 2019-07-12 by htcohio