OpenHD/OpenHD #86: ***IMPORTANT*** Ground Video feed lost during Air power off or Out-Of-Range/Doesn't Come Back

htcohio, 2019-06-03:
As stated, it has been reported a couple times that either sometimes or under certain circumstances when the Drone is out of range or it's power-cycled, the video does not come back.

1. It is possible this is more common with RTL8812 + Encrypted RC.  
However, it seems like this has happened with Range RC at least once.

https://www.rcgroups.com/forums/showthread.php?p=41888061

pilotnbr1, 2019-07-01:
This appears to be related to certain frequencies on 5.8 dfs and radar...

htcohio, 2019-07-12:
This likely relates to DFS channels getting out of the way of radar even though it's disabled in the source code.

Best to stick with 5.8 channels 36, 38, 40 & 42 also ch 149-165 seem ok

closed 2019-07-12 by htcohio