OpenHD/OpenHD #88: Fix the support for TPLink WN822N

PetruSoroaga, 2019-06-07:
This card (TPLink WN822N) was working in EzWifiBr... 1.6.
Not working with OpenHD (I thing the last image available from April or something).
If and when you get a chance, It would be great if you can have a look and see what broke (or add support for it).
Might not be so common, but on my area it's the only one I have access to (besides the classic  722 one)
Thank you,
Petru

htcohio, 2019-06-08:
@PetruSoroaga 

Thanks again for bringing this up. 

It seems like there is an additional patch to support the WN822N dual band Atheros cards... 
It must have been overlooked. 
http://fpv-treff.de/viewtopic.php?f=23&t=8095

raphaelscholle, 2022-10-16:
since the drivers are quite old and not maintained anymore we'll propably never add it in newer versions

closed 2022-10-16 by raphaelscholle