OpenHD/OpenHD #97: Two Gpio-config.py

pilotnbr1, 2019-06-16:
We have two gpio-config.py files now due to a recent fix. 

New gpio-config.py with the "working" fix is in the script repo here now

Old gpio-config.py is still being added in the builder stage 4. It also has a service that is related to gpio funtion. Not sure if it is needed or not.

This post is essentially a to-do cleanup reminder so its not forgotten.

htcohio, 2019-07-12:
This has been resolved

closed 2019-07-12 by htcohio

pilotnbr1, 2019-07-12:
I see one added here by Petru
https://github.com/HD-Fpv/Open.HD/blob/development/wifibroadcast-scripts/gpio-config.py

And one here that is added by the image builder
https://github.com/HD-Fpv/Open.HD_Image_Builder/blob/development/stages/04-Wifibroadcast/FILES/overlay/root/wifibroadcast_misc/gpio-config.py

Here the one-shot service calls wbconfig.sh which references the original (one added by image builder)
https://github.com/HD-Fpv/Open.HD_Image_Builder/blob/development/stages/04-Wifibroadcast/FILES/overlay/root/wifibroadcast_misc/wbcconfig.sh

I asked Petru to reconcile this difference but no response via rcgroups as this was his change. Ill take a stab at this and delete Petrus gpio file. They look Identical...

htcohio, 2019-07-13:
Hey,

Vitaliy rewrote and corrected any problems. Its been working perfectly.

Should we change anything?

pilotnbr1, 2019-07-13:
Yes... even though its working, anyone who comes along later that tries to figure out or modify gpio stuff is going to be super confused