OpenHD/OpenHD #100: PiP: No killpattern for low datarate Secondary video stream

careyer, 2019-06-18:
Now that we have a working Picture-in-Picture (PiP) prototype the following observation can be made: 

**Observation:**
If the secondary videostream has a very low datarate (e.g. FLIR: 160x128 9fps) the video only gets  displayed with a huge delay after activating the secondary videostream. The delay can reach up to 20sec before the video finally shows up in the QGC app. 
[This behavior can also be reproduced  with the videotestsrc gestreamer pipeline with similar resolution and framerate settings. (Here the delay will be a bit shorter due to much more movement in the video and hence sligthly more datarate - however delay will still be in the 5-10sec range).]
In our tests we could also observe that the delay will not be that bad if the videostream gets not killed completely when toggeling the secondary video off.  When the secondary video pipeline stays alive it can then immediatly be used once the secondary videostream gets activated again. 

The assumption is that when starting the gestreamer videopipeline some buffer needs to be filled first before the video gets computed/transmitted:

**Possible solution:**
For low datarate streams negativ effect on the bandwidth can be neglected and such streams need not to be killed but their videopipelines (maybe also tx_rawsock instances) can stay alive. Just don't show the video at GroundPi/QGC. This way delay when activating secondary video can be reduced to a minium.

My suggestion is to introduce a config setting such as:
`Keep2ndVidStreamAlive=Y/N` (the according comment should state that this is only suiteable for low datarate streams)

Thanks!

user1321, 2019-07-28:
Hi.
Do you still able to notice that delay even after that param was added:
omxh264enc control-rate=1 target-bitrate=600000 interval-intraframes=1 periodicty-idr=1
to gst-launch program?

careyer, 2019-07-29:
@user1321 : Sorry I totally forgot about this issue. Thanks for bringing this to my attention again!

The delay issues was indeed solved by the additional OMX parameters. The delay was obviously caused by the lack of key-/idr-frames. All working smoothly and as expected not! Hence closing this issue now!

closed 2019-07-29 by careyer