OpenHD/OpenHD #99: Ground PI stuck in smart synchronization(ar9271 and rtl8812au)

zipray, 2019-06-18:
Simultaneous insertion of rtl8812au card and ar9271 card on the ground will be stuck in smart synchronization.

HD-Fpv, 2019-06-19:
so just to clarify, are you saying that ar9271 + 8812au simultaneously connected on the ground cause it to freeze up?

htcohio, 2019-06-19:
Hey Zipray,

in regards to this issue.

I put this video together and I had 722N +8812AU on the ground side and as you can see everything boots up fine on my end.

*this is with the image that was compiled on June 10th... 
https://youtu.be/oynntEEyUgA

zipray, 2019-06-19:
> Hey Zipray,
> in regards to this issue.
> I put this video together and I had 722N +8812AU on the ground side and as you can see everything boots up fine on my end.
> *this is with the image that was compiled on June 10th...
> https://youtu.be/oynntEEyUgA
Thx Justin
Where is the image that was compiled on June 10th?

Beegad42, 2019-06-25:
The symptom is that the boot process halts with a picture of a Taranis RC top right and the text "SSync screen started" bottom left.

I am battling to get Open.HD running and have tried a variety of cards and have yet to get much success with Open.HD-2019-06-07_Post_Master-Merging.img.zip. Have had some success with EZ-Wifibroadcast-1.6RC6+8812au.zip (I am a newb to both).

This issue of the boot process halting seems to be related to Wifi card selection. As I remember it can be reproduced by plugging Realtek versions of TP-Link 722 or 822 USB cards (i.e. the later versions or those adapaters).

I am having similar issues with ASUS AC56 (8812AU?) on 5.8 GHZ, trying to put the hotspot on internal (3B+) 2.4 Wifi causes Open.HD to stall in the boot process as reported by OP.

htcohio, 2019-06-25:
hi, you must be using an older image before we are ironed out all of the
proper settings for smart sync.

today I'll be loading a new image to the wiki with a brand new settings
layout as well as smart sync disabled by default.

you can still use smart sync but you just need to install a simple switch
or connect your joystick via USB to the ground RPI.

user1321, 2019-07-14:
Hi.
If question was in:
1. set 5.8 GHz.
2. Insert 722N +8812AU  = result hang.
Than, it is fixed by skipping ath9k configuration error system halt (it can`t be configured using 5.8GHz)

Also, it will boot even using singe wifi card and wrong configuration:
722N - 5.8 GHz
8812AU - 2.3 GHz
But, that don`t meen it will work. It don`t. just boot in order to resolve problem using android app (set correct freq)
Please, reopen if problem not solved for you.

https://github.com/HD-Fpv/Open.HD/commit/4e53322bfedb95ac1f0057178928274c3844d72d#diff-9ff9316bfa6c35330e6fcad2a82d9411

+

https://github.com/HD-Fpv/Open.HD/commit/f723b00dc3675b2e2207c2f521b49d1d45a71ccc#diff-9ff9316bfa6c35330e6fcad2a82d9411

closed 2019-07-14 by user1321