OpenHD/OpenHD #101: Proposition: LED heartbeat

zygmunt, 2019-06-22:
Hi,

I started playing with OpenHD (Im new here) after some long break from wifibroadcast at all.
Im not sure that my RpiZero works, so I want to propose add LED blinking:
For example:
https://mikebeach.org/2015/11/01/raspberry-pi-heartbeat-led/

PS: I see that LED blinks on rpi v2, but Im not sure why (maybe it is OpenHD behavior, maybe bot)

Regards
Zygmunt

pilotnbr1, 2019-06-22:
The led functionality has been intentionally patched to disable the led blinking. On some wifi cards that blinking was found to cause packet loss. Bizarre I know...

htcohio, 2019-07-14:
Yeah, I'm closing this issue, the factory LEDs are totally useless outside. 

Maybe in the future we can add an option for a gpio external LED?

Just trying to clean up and prioritize critical issues at the moment.

Might revisit this in the future.

closed 2019-07-14 by htcohio