OpenHD/OpenHD #102: RaspberyPi Zero not working?

zygmunt, 2019-06-22:
Hi,

Can anyone confirm that current version of OpenHD works on Rpi0?
Im using USB TL-WN722N, with USB OTG.

The same SD card works fine on RpiV2, but I dont see a video signal from Rpi0.

PS: I tried login into rpi0 over ssh and serial port, seems to works, but after some time looks like freezed. Green LED on wifi cards is ON, looks good, and Rpi0 stays hot.

Regards
Zygmunt

HD-Fpv, 2019-06-23:
Zygmunt,

sorry you seem to be having troubles combined with pi zero, pi 1, and the wnd

pilotnbr1, 2019-06-23:
Openhd confirmed working on pi0... Alot of us use it as an airpi. 

We need more info to help you.. Does it boot? If so what messages do you see? 

Please enable debug=y in the settings file and share logs.

Most common problem is power. Make sure it has proper voltage and amps.

zygmunt, 2019-06-23:
Hi,
lets focus here on RpiZero and TL-WN722N (which works good with RPIv2, and the same SD cards) :)

> Does it boot?
Yes, but after some time it looks like frozen.
I didn't sloder anything on rpi0, first I started testing everything.

pilotnbr1, 2019-07-01:
pi zero works...

closed 2019-07-01 by pilotnbr1