OpenHD/OpenHD #103: The ground use external wifi card to do hotspots can not work

zipray, 2019-06-22:
The ground use external wifi card to do hotspots can not work, enable remote settings can find hotspots but can not get ip, disable remote settings can not find hotspots.V6.19

zipray, 2019-06-25:
Latest test:
1. The ground adopts ar9721 card as rx card, with working frequency of 2.4g and rtl8812au as external hot spot card.
A. Set the hotspot at 5.8g and it works normally.
B. Set the 2.4g frequency band hotspot. The external hotspot on the ground does not appear to be activated and cannot be detected by ssid.
2. The ground adopts rtl8812au as the RX card with frequency 5.8g, and ar9271 card is set as external hotspot  2.4g. The ssid of ground hotspot can be detected,.
A. the mobile phone enables DHCP to try to connect to the external hotspot , it cannot obtain IP, which results in the termination and exit of the external hotspot.
B. The mobile phone set static IP 192.168.2.2, and the gateway address 192.168.2.1 is connected successfully, but the remote setting application cannot obtain parameters, and there is no video image.

htcohio, 2019-07-01:
Zip,

I was just able to also confirm the same problem with external hotspot...  Sorry it took so long... 

I'm sure this is something that Luke can fix fairly easily..

Luke,

any combination of external "Yes" "no" or "Auto".... + regardless of if the old AP config text file mattress or not the external Wi-Fi card still tries to boot up for the main openhd video...

pilotnbr1, 2019-07-01:
The external hotspot and nic support was completely dropped along the way. A fix has been added to dev branch.

zipray, 2019-07-01:
> The external hotspot and nic support was completely dropped along the way. A fix has been added to dev branch. Currently the fix needs testing.
Thanks for your fix. Although I haven't tested it, is the following code repeated setting normal?

https://github.com/HD-Fpv/Open.HD/blob/2735171b104a0d6b58184a262bdb5d5ac8069283/RemoteSettings/Ground/helper/ConfigureNics.sh#L233

https://github.com/HD-Fpv/Open.HD/blob/2735171b104a0d6b58184a262bdb5d5ac8069283/wifibroadcast-scripts/hotspot_functions.sh#L21

pilotnbr1, 2019-07-04:
Closing. It works now

pilotnbr1, 2019-07-07:
Just to document this further. With ac56 and or 722n I can create ext hotspots. 
Ac56 hotspot you can connect to on 5.8 and 2.4

With 722n the created hotspot does not allow connection. On my android device it says failed to get ip address

Needs more debugging...

raphaelscholle, 2022-10-16:
Hotspot will be made with network-manager and not with openhd, issue should be resolved

closed 2022-10-16 by raphaelscholle