OpenHD/OpenHD #108: USB tethering Broken

zipray, 2019-07-07:
OpenHD image July 5 version：[OpenHD image download
](https://drive.google.com/file/d/19j73pvlBmSMDVMZ8ynQu2ACqVgf2uydW/view?usp=sharing)
"ERROR: Could not configure IP for USB tethering device!" 
The spread of  #103 ？

htcohio, 2019-07-08:
Hi,

I actually noticed that today also.  Can you say that this is something new? Or was USB working previously?

Im also curious, do you have a 3 B+ that you use?

htcohio, 2019-07-12:
Closing this,

My phone turned out to have faulty USB. 

Others confirm working well.

Notes if having problems: 
1: USB tethering does not work if VPN is enabled.
2: USB tethering sometimes does not work if the phone's power save mode is activated.
3: Disable any firewall or third-party antivirus software that might be running.

closed 2019-07-12 by htcohio