OpenHD/OpenHD #114: Switch only shows IP cameras must have a screen connected to HDMI

zipray, 2019-07-12:
This is one of the strangest situations where you can't switch to the (only)  IP camera mode without connecting to the hdmi monitor.However, the PI camera and IP camera  (together) display can be switched, whether or not the HDMI monitor is connected.Also, USB cameras are not affected by this problem.Whether or not the ground PI is connected to an HDMI display, it can switch between three states.The only problem is that sometimes the switch fails, which is may related to if success of the HDMI display decoding.

user1321, 2019-07-14:
Hi.
I am sorry, can you please confirm that I understood correctly:

RPi Air:
RPi <-> RPi camera
RPi <-> Ethernet IP camera
RPi <-> USB 8812AU (or anything else)

RPi Ground:
RPi <-> USB 8812AU
RPi <internel hotspot> <-> Android phone

as result you can`t display at Android phone video from IP camera?

But, if you will change to that configuration:

RPi Air:
RPi <-> RPi camera
RPi <-> USB camera
RPi <-> USB 8812AU (or anything else)

RPi Ground:
RPi <-> USB 8812AU
RPi <internel hotspot> <-> Android phone
_________________________________________
You can, Is it correct?

zipray, 2019-07-15:
RPi Air:
RPi <-> RPi camera
RPi <-> USB camera
RPi <-> USB 8812AU 
1、
RPi Ground:The toggle switch is in third gear.Android app rtp port 5601.
RPi <-> USB 8812AU
RPi <-> Android phone+HDMI monitor
The toggle switch is in third gear.Android app rtp port：5601.
Everything is all right
2、RPi Ground:The toggle switch is in third gear.Android app rtp port 5601.
RPi <-> USB 8812AU
RPi <-> Android phone（without HDMI monitor）
I can't switch to the  IP camera (only) mode.The switch on  third position.
but I can switch to the  IP camera on  Android phone mode.The switch is on  second place.

user1321, 2019-07-15:
Got it.
I will try to reproduce and let you know.
Thank you.

closed 2019-07-29 by zipray

VU2ATX, 2019-09-16:
> 1、
> RPi Air:
> RPi <-> RPi camera
> RPi <-> Ethernet IP camera
> RPi <-> USB 8812AU
> RPi Ground:The toggle switch is in third gear.Android app rtp port 5601.
> RPi <-> USB 8812AU
> RPi <-> Android phone+HDMI monitor
> Everything is all right
> 2、
> RPi Air:
> RPi <-> RPi camera
> RPi <-> Ethernet IP camera
> RPi <-> USB 8812AU
> RPi Ground:The toggle switch is in third gear.Android app rtp port 5601.
> RPi <-> USB 8812AU
> RPi <-> Android phone（without HDMI monitor）
> I can't switch to the IP camera (only) mode.The switch on third position.
> but I can switch to the IP camera on Android phone mode.The switch is on second place.
> 3、
> RPi Air:
> RPi <-> RPi camera
> RPi <-> USB camera
> RPi <-> USB 8812AU
> RPi Ground:The toggle switch is in third gear.Android app rtp port 5601.
> RPi <-> USB 8812AU
> RPi <-> Android phone（without HDMI monitor）
> Everything is all right

Hi, I am following the wiki [To test camera switch (RPI native camera + virtual USB\IP)](https://github.com/HD-Fpv/Open.HD/wiki/Software-~-Advanced-~-USB).

[image]

My Setup:
**RPi Air:** 
RPi: Raspberry Pi 2B
CSI2 Camera: PiCam 5mp, V1
USB Camera: Logitech C920
WiFi USB Adapter: AR9271L based adapter with external antenna.

**GroundPi:**
RPi: Raspberry Pi 3B+
WiFi USB Adapter: AR9271L based adapter with external antenna.
**Display:** 
    1. Android Phone connected via USB teathering for Video / Telemetry QGroundControl apk 
       installed.
    2. HDMI cable also ready to be connected to the HDMI monitor.
Joystick: Transis X9D connected via USB Cable. Ch7 is ready with PWM min=1000 and PWM max=1999 to be used to switch camera.

I am not getting to what you said : Toggle switch in third gear, there is nothing about it on wiki, I am refering to.

I am not getting success in camera switch however I am getting video when using single USB camera.
Waiting to hear from you.