OpenHD/OpenHD #117: IsBandSwicherEnabled=1 breaks Ethernet/SSH on AirPi

careyer, 2019-07-16:
Hi VItaliy,

in a long debug session last night Justin and me found a strange behavior which is caused if the "IsBandSwicherEnabled=1" setting is used. (BTW: There is a typo in the name: swi(t)cher, but that is only a side-node here, you might want to fix that just for esthetics  ;-). 

Our Test setup was:
- AirPi: Pi3B
- WiFi-NIC: TP-Link 772N
- Ethernet NIC connected to Home-Router (DHCP enabled)
- Image: Open.HD-2019-07-05.img
- All preconditions met to SSH into the AirPi (i.e. ssh file in /boot, DEBUG=Y)
- Setting hunted down to cause the problem: "IsBandSwicherEnabled=1"

Observation:
If "IsBandSwicherEnabled=1" in enabled the Ethernet NIC most of the time does not get an IP assigned by the rounter. Only in very rare cases (which involved restarting the router) an IP was assigned but right on the very next reboot of AirPi that did not work anymore. 
Hence SSH into the AirPi is not possible. When deactivating/setting "IsBandSwicherEnabled=0" the IP assignment will work reliably again on every boot and SSH is possible.

Desired behavior:
Regardless of "IsBandSwicherEnabled=1" or "IsBandSwicherEnabled=0" the Ethernet interface should reliable come up and receive IP via DHCP. SSH should be possible in order to support propper testing/developing tasks.

user1321, 2019-07-27:
Hi, two days ago I added to RPi Air side simple filter that will configure Eth0 interface only in case if Secondary camera set to IP camera.
Hope that will help.

https://github.com/HD-Fpv/Open.HD/blob/singlecamera/RemoteSettings/BandSwitcherAir.sh

htcohio, 2019-09-05:
Confirmed fixed

closed 2019-09-05 by htcohio