OpenHD/OpenHD #120: 2-Way Audio Help

gurustave, 2019-08-02:
I am part of a group working on using Open.HD for an RC Ground Vehicle. I know Open.HD Supports One-way audio from the "air" pi to "ground" pi, but we'd like to also be able to transmit audio over this system to the "air" pi as well. I'm fairly comfortable with BASH, but quite a bit more rusty in C. Any help, suggestions, ect. would be much appreciated.

pilotnbr1, 2019-08-02:
There’s a lot of bash so you should be ok!

Here at line 261 you can see where audio gets started up 
https://github.com/HD-Fpv/Open.HD/blob/master/wifibroadcast-scripts/tx_functions.sh

You will need to follow it and set it up going the other way too.

gurustave, 2019-08-10:
Thank You @pilotnbr1. Showing me where to start was a great help in getting this functioning. If anyone is curious, you can see the changes I made to get this working here. https://github.com/frc3468/Open.HD/commit/9fdcf9829204f44ef1985d129938e4aa92e26e64

closed 2019-08-10 by gurustave

corteXnzt, 2021-04-22:
I'm trying this now, I know this is still from 2019 but just in case you can still recall. I have updated the paths and copied your ports and radio channel but the Ground stations puts on hold mode and control freezes when I use this scripts. Have you encountered this during your development?

gurustave, 2021-04-22:
I never got it working, but as I remember, my issue was that if I had both the microphone and joystick plugged in no data transmitted. If I had only the Gamepad on boot, it transmitted controls just fine, and similarly if I had the usb microphone plugged in and no Gamepad it worked just fine. I never got around to investigating further and at this point was just going to try again with the new greatly improved version of the software.

corteXnzt, 2021-04-22:
Thank you for fast reply. Maybe we have the same issue since I'm using a usb custom made joystick and the video feed is still working. I'll look on this. Thanks again.