OpenHD/OpenHD #121: joystick buttons

Howie9600, 2019-08-04:
is there a way to define joystick buttons?

Thanks

Howie9600, 2019-08-05:
anyone?

gurustave, 2019-08-10:
What do you mean by define them?
Joystick Buttons are already captured and transmitted to the Air Pi. I believe the only RC protocol that transfers the switches to the Drone Controller is Mavlink, at least as the code is currently written. 

I personally modified the 8th Axis for the MSP Protocol in rx_rc_telemetry.c to transmit the switches instead as I am using an Arduino to control my actuators and can parse the data however I like.

Howie9600, 2019-08-15:
so the joystick file defines axis but not buttons.  I would like to define buttons instead of axis over mavlink

Howie9600, 2019-08-16:
how do i assign JSSWITCH 1 to channel 5 for example to control flight modes

gurustave, 2019-08-16:
I don't see any method built in for assigning the buttons, but it looks like for mavlink, the first 10 buttons are sent as channels 9-18, with a value of 2000 if it is pressed, and 1000 if it is not pressed.

mjc-506, 2019-09-03:
Yes, currently the buttons are sent just as 'channels', so 2000 if on/pressed, 1000 otherwise.

I made some changes to rctx.c on the groundpi (ssh in) to make the buttons changes a particular rc channel, which the FC then uses for flightmode.

I've attached an example - this likely won't match your requirements exactly, but is hopefully a starting point. I suggest ssh into the ground pi (you will need DEBUG=Y in the config file) and play from there!
[rctx.c.txt](https://github.com/HD-Fpv/Open.HD/files/3570214/rctx.c.txt)

Howie9600, 2019-11-07:
Ok I have solved this issue and now all my flight modes on buttons and gimbal control on buttons. This only works with mavlink

closed 2019-11-07 by Howie9600

pilotnbr1, 2019-11-08:
> Ok I have solved this issue and now all my flight modes on buttons and gimbal control on buttons. This only works with mavlink
> 
> If anyone is interested

@Howie9600 yes would be nice to see your solution!

Can you share your relevant code here please!

Howie9600, 2019-11-13:
It wasn't code for the actual software.  I built a joystick and coded the joystick to do it and kept arducopter the same.