OpenHD/OpenHD #123: Surface Pro>Ethernet Hotspot>GStreamer no display after command prompt

cantfaxglitter, 2019-08-15:
Attempted to connect using GStreamer in Command Prompt.

TX: PiZero w/Zero4U USB Hub + Panda Wireless PAU09 + Pi Camera V2
RX: Pi 3B+ w/Panda Wireless PAU09
GStreamer: (Via Open.HD wiki link) Tried GStreamer 10.2 32 bit, then 10.2 64 bit
Added firewall outbound/inbound exceptions for  gst-launch-1.0.exe 32/64

At Command Prompt:
$ gst-launch-1.0.exe udpsrc port=5600 ! h264parse ! avdec_h264 ! autovideosink sync=false

Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock

At Command Prompt:
$ gst-launch-1.0 videotestsrc ! videoconvert ! autovideosink

Correctly displays GStreamer Test Window

OpenHD Android App works great, connected via Wifi Hotspot. 

Frequency: 2472
Wifi NIC: auto
Wifi Hotspot NIC: internal
E. hotspot: Y
Forward: raw
Video Port: 5600

HDMI from RX Pi 3B+ works great at startup on TV display with minimum latency, just nothing on Surface Pro

Was able to SSH (via Putty) into Pi3B+ successfully.

Howie9600, 2019-08-15:
did you allow ping through the firewall?  The code pings the pc and then starts the stream.  You need to allow ping through windows firewall (or any other you are using)  Once you allow this the code pings the pc ip address and starts the stream

cantfaxglitter, 2019-08-15:
Yeeup, so I think that was the problem. Turned off the firewall and the stream worked perfectly via Ethernet hotspot. 

I’ll keep looking around how to make an exception for windows 10 to allow the ping through the firewall then document here and close the issue.

Howie9600, 2019-08-15:
It caught us for a while so i was forcing the ping bit until i found it easier to add the rule as below

Search for Windows Firewall, and click to open it.
Click Advanced Settings on the left.
From the left pane of the resulting window, click Inbound Rules.
In the right pane, find the rules titled File and Printer Sharing (Echo Request - ICMPv4-In).
Right-click each rule and choose Enable Rule.

cantfaxglitter, 2019-08-24:
Yeeup Enable Rule works for my setup, closing issue.

closed 2019-08-24 by cantfaxglitter