OpenHD/OpenHD #124: How to control Pixhawk usb joystick

naughtyxp, 2019-08-25:
Hi
How do  control pixhawk with usb  joystick.

i connect joystick with groudnpi usb port

pilotnbr1, 2019-08-25:
Make sure you have 2 way mavlink telemetry working. In settings be sure under “rc” you have entered “mavlink”

naughtyxp, 2019-08-26:
Hi,

RC is mavlink ok but notworkin :D 

> Make sure you have 2 way mavlink telemetry working. In settings be sure under “rc” you have entered “mavlink”
> 
> If your joystick is recognized by the pi you should be good to go...

Hi,

[image]

[image]

pilotnbr1, 2019-08-26:
Have you configured stream rates for the pixhawk?

https://github.com/HD-Fpv/Open.HD/wiki/RC-~-RC-with-Ardupilot

pilotnbr1, 2019-08-27:
Cool! Those stream rates burn everyone.

Closing....

closed 2019-08-27 by pilotnbr1

naughtyxp, 2019-08-27:
> Have you configured stream rates for the pixhawk?
> 
> https://github.com/HD-Fpv/Open.HD/wiki/RC-~-RC-with-Ardupilot

[image]

naughtyxp, 2019-08-27:
its worked :D sidewinder joystick control groundpi :D