OpenHD/OpenHD #128: Git workflow

LimitlessGreen, 2019-08-29:
Hello,
it's nice to see a repo where the work of EZ-WifiBroadcast goes future :+1: 
I noticed some things that should be improved. Especially because other developers should better understand what's going on here. I'm not sure. Are you new to git?

- **The funniest thing I noticed:** mostly all of your commits are GitHub internal gpg signed. Conversely, that means you're using the web frontend for all the stuff here. There is no reason not to do that. Can you tell me a little bit more about that? Are you using the web editor or drag and drop files into GitHub?
- **Next, your image builder:** The content looks like RespawnDespair's image builder. But technically it isn't. Instead, I see copy-paste commits like this: https://github.com/HD-Fpv/Open.HD_Image_Builder/commit/b9f27854fd5d58403130ad5349e7d2f0d1662e1d
Why not just do a simple fork?
If you want I can help you to clean this up. I have the advanced git knowledge to do that. 
And the end you should have a clean repo forked from @RespawnDespair or @user1321. And the best thing: It is comparable now! That would be extremely helpful to other developers. It's hard enought to look through the jungle between forks and developments of EZ-WifiBroadcast.
- **Last one, [this repo](https://github.com/HD-Fpv/Open.HD):** I really liked the modularity of RespawnDespair's repo. But here all the parts again smashed together. Why? From the developer's view, it's great to import only some modules of a big project instead of the whole thing. And sadly again I see copy-paste commits in this repo. https://github.com/HD-Fpv/Open.HD/commit/d78209f4bef7dd9b9e1a2dd205c46ccb872da32e
My advice to you to keep things clean (and again of course I can make this for you):
All of the modules you "imported" into this repo should be a fork in this account. If you want to improve things, simply push the changes to the forked repos. And here you should import these forks with git submodules. This is a simple and really clean way. :smiley: :sparkles: 
---
I hope you accept my help :octocat: 

Best regards,
Green

htcohio, 2019-08-30:
@LimitlessGreen, 

Thank you for reaching out!  First, I myself am not a coder, I would probably be considered the lead tester of the system, it has a lot of moving parts and adding stuff does have a tendency to break other things from time to time.

 There is a lot of background and people involved where we created a brand new account "HD-FPV" Where we all have some control. 

The last year has been fast and loose development continually adding features, Most of the "New-Original Code" Was added by USER1321 who has added some really awesome features combined with 1.6 RC6 stability, 1.6 RC6 really was a difficult benchmark to reach.

I could go on and on with info but, I would like to leave space here for Respawndespair, User1321, Pilotnbr1 (Luke) who can speak about the workflow  better than myself.

FYI, Every day there is ongoing discussion and recently very similar to what you are saying.

We use the Public telegram group (see link).  It is our primary means of communication because a lot of the system requires sharing pictures video and files easily.  I think we have nearly 110 people on the group so if you are willing to join in on there we would welcome any ideas you may have.

Https://t.me@OpenHD_May_2019

OpenHD 2019_RCGroups Telegram Group
This Open.HD group is intended to replace or supplement the Open.HD RCGroups Forum which is outdated and difficult to share media on. 

Public Group
https://t.me/OpenHD_May_2019

From there, There are also some smaller Developmemt-Oriented discussions as well. 

Basically,, every single day

LimitlessGreen, 2019-08-31:
@htcohio 
Thank you for your answer! It helps me a lot to understand how all things are connected.
>  it has a lot of moving parts and adding stuff does have a tendency to break other things from time to time.

That's a big thing I noticed while reading through the lines of code and of cause the problem was also touched in the Telegram group.
Do you know how big projects like the Linux Kernel stay maintainable? 
They use well-defined interfaces, abstraction layers ... and very important ... they are **highly** modular. Engineers often go to the drawing board and share ideas before adding stuff into it. The diagrams in your developer's corner are a great step in the right direction :+1:
At this point, it looks like you spend a lot more time into bug fixing than developing. Am I right?
It's time to make the necessary changes to the code base otherwise the project will become unmaintainable very quickly.
Do you know the hell's curve rise (worst-case) of unmaintainability? With only 2 related modules the rise of bug-fixing could be not linear, not exponential...it could be infinite!
And trust me, making the changes now will save a lot of time in the future.

...But that wasn't really the topic of this issue :see_no_evil: 

Btw, it should be nice to hear from @RespawnDespair or @user1321 or @pilotnbr1 something about the git part.

@htcohio again:
I'm not sure if you understood 100% what I wanted to say :sweat_smile:
My Issue was about to change the Git-base, not the Code-base (for now).

I'll give you a quick example:
Sometimes a developer wants to know what the idea or reason behind a single line of code was. A great function to do this is `git blame`. And what do you want here is the description line of the original commit. But what you **get in this repo**, are things like **`Merging all sources here`**. This isn't helpful for developers. Did you get the glue? :smile: 
It's possible to recover this. And here is the point I want to help :octocat: 

---
Best regards,
Green

htcohio, 2019-08-31:
> @htcohio again I'm not sure if you understood 100% what I wanted to say sweat_smile
My Issue was about to change the Git-base, not the Code-base (for now). 

Again thank you for your inquiry and int I think your ideas are well accepted but also a little intimidating but,  could help us all understand best practices.  

I should be able to provide any information or feedback related to the why and how various features and enhancements that User1321 in particular added tho OpenHD and Builder over the last 6 months or so.  I would imagine that some of the script that is related to things like SmartSync or other Hardware items might not seem clear right away.  I am pretty sure User1321 has a lot going on right now outside of his project so his availability to help might be somewhat limited at the moment.
I'm sure you have read that @pilotnbr1, @faceplant8  (BrianWebb), @RespawnDespair have also suggested cleaning up the project a bit over the last week. 

Actually it seems like open HD has started to get the attention of more developers which is pretty cool to see. Overall, OpenHD is very dependent on the Raspberry Pi (E.g. Raspivid, Hello_Video, Innomaker MIPI Cam, Arducam switcher, GPIO stuff etc... Some people have expressed interest in the other linux SBC's for flexibility but also because Raspberry Pi workflow has somewhatb of a challenging "work-flow.". 

With that said, it looks like you've joined the telegram group? So maybe we can just make another message group on Telegram?   

From there,  What are you thinking the first steps should be? And, is there anything in particular that myself or others would be able to better explain?

P.S. You may notice that our Wiki was originally forked from the Respawndespair Imagebuilder/Wifibroadcast and some of the Wiki items are outdated and need to be updated.... So, although that is always a work in progress, most of the core usage and functionality is correct.

Kind Regards,

htcohio

LimitlessGreen, 2019-08-31:
> Again thank you for your inquiry and int I think your ideas are well accepted but also a little intimidating but, could help us all understand best practices.

You got the point 👍
And yes of cause changes are often intimidating.

> I should be able to provide any information or feedback related to the why and how various features ... 

Great! this would help a lot.

> I'm sure you have read that @pilotnbr1, @faceplant8 (BrianWebb), @RespawnDespair have also suggested cleaning up the project a bit over the last week.

I'm not sure that I found the right section in the history 😅 

> Actually it seems like open HD has started to get the attention of more developers which is pretty cool ...

Of course, that are great news for a project like this 😄
This dependence on the Raspberry can be lifted (but that's a future goal). ...That's where modularity and abstraction layers comes in place. One recommended thing to do this is to outsource functionality into devices. For example, speak with /dev/servo0 instead of doing GPIO stuff. 

> With that said, it looks like you've joined the telegram group? So maybe we can just make another message group on Telegram?

That's right 👍 To open up a new Telegram group is a very good idea. Just search for my icon. 

> From there, What are you thinking the first steps should be? And, is there anything in particular that myself or others would be able to better explain?

1. Fork the image builder from @RespawnDespair 
2. Merge (pls do not understand this as a git merge...it's a bit more of magic) the commits of OpenHD's image builder step by step onto it (and don't forget to pick @user1321 commits too). Here I need your help to resolve the "Merged from..." commits.
3. ... I'll explain that if the time comes

---

Another little thing this way. It should be great if a little hint was added to for example EZ-WB that this project exists. Other users (included me) thought the project was dead.

---
For now:

Best regards,
Green

RespawnDespair, 2019-09-01:
As posted on Telegram, i am very interested in cleaning up the repository and moving forward with the code base in a manner that allows new devs to more easily adapt. I would suggest joining the telegram group if people have not done so already. From there we could try to set up a voice chat sometime soon. It will probably take some scheduling since everybody is in a different timezone, but we should be able to work something out.