OpenHD/OpenHD #129: Git Cleanup

LimitlessGreen, 2019-09-01:
# Related to issue #128, this is the process history for the cleanup.

## Goals:

## Things To do:
- [ ] fork https://github.com/RespawnDespair/wifibroadcast-image-builder.git
- [ ] Add remotes: @HD-Fpv @user1321 
- [ ] Add a new branch `cleanup` to @RespawnDespair master
- [ ] reset `cleanup` to 64f4d2ecb094c1910b20eb095c72bf9ae7b8badc cause the last 2 commits are already in the fork of @user1321 
- [ ] rebase @user1321 master above commit 5712979c9c2b5bde26688e907c8a6c265cba7012 to b0c499ff7a2d084a892fdea58d466eff513e72ac onto `cleanup`  (that's one before the master pointer. Is the last commit needed?

## Command history
- `git init`
- `git remote add [fork] [fork URL]`
- `git remote add  RespawnDespair https://github.com/RespawnDespair/wifibroadcast-image-builder.git`
- `git remote add user1321 https://github.com/user1321/wifibroadcast-image-builder.git
`
- `git checkout RespawnDespair/master`
- `git checkout -b cleanup`
- `git reset --hard 64f4d2ecb094c1910b20eb095c72bf9ae7b8badc`

pilotnbr1, 2019-09-02:
Nice to have a git expert willing to help! Speaking for myself I am terrible with git...

Beegad42, 2019-10-09:
the file settings.sh should be deleted https://github.com/HD-Fpv/Open.HD/blob/master/wifibroadcast-scripts/settings.sh

Beegad42, 2019-10-09:
the directory wifibroadcast-rc-encrypted does not appear anywhere in the scripts and therefore can be deleted. All the functionality was moved to wifibroadcast-rc-Ath9k

pilotnbr1, 2019-10-10:
Wifibroadcast/hello-video contents aren’t just for looks... the programs actually doing the work are in opt/vc/src/hello pi/hello video (or close to that)

RespawnDespair, 2020-12-04:
Closing this as we are quite past this point now 😄

closed 2020-12-04 by RespawnDespair