OpenHD/OpenHD #134: Interprocess Communication

RespawnDespair, 2019-09-03:
Currently the different C/C++ classes use SHM to communicate between different applications. 
In the refactor we want to have a unified way to do the following across all applications:

- Read status information 
- Log messages on different levels to a central logging mechanism
- In some instances write updated configuration

RespawnDespair, 2019-09-03:
I like the svpcom approach (https://github.com/svpcom/wifibroadcast/blob/master/telemetry/cli.py) of using UDP ports that allow, in potential, bidirectional communication.
It also allows multiple programs to read the stream of information. So in my view UDP could cover points [1] and [3].
As for point [2] the reactor logging in Python seems to do it's job fine when running as a service through systemd where i can retrieve the logs as for any other service. I do feel there might be a more elegant way, but it does seem like the linux way of doing this.

webbbn, 2019-09-04:
I think you've conflated two problems: logging and configuration updates.

The standard Unix/Linux model is to use syslog for logging, and I believe that will far exceed the requirements.

The standard for configuration is typically text files that are read on startup, and (if necessary), you can sometimes signal a process to re-read it's configuration file(s) by sending it a SIGHUP.

RespawnDespair, 2019-09-05:
Well, yes and no, i tried to separate pure logging from realtime status. 
Apart from that we have the INITIAL config which offcourse should be stored in a config file. (I like the INI file format). But we also have some realtime configuration changes that are possible in the system. E.G. dynamic camera setting changes. There should be a protocol in place for that as well...

webbbn, 2019-09-05:
Can you give an example of dynamic camera setting changes? What parameters are being changed? Should those changes be permanent (written to a config file)?

It would be nice to have a list of current and desired functionality that we can refer to.

RespawnDespair, 2019-09-06:
> It would be nice to have a list of current and desired functionality that we can refer to.

Yes that would be nice, i can't give you an example, i know from past discussion that people have worked on being able to change the exposure etc from the picam on the fly. As we are redesigning i figured that should be included in the redesign. It might qualify as new functionality.

I figured it might have an impact on the design choices we make now... But to not stray from the point:

[1] Sockets of some form over sharedmem?
[2] syslog
[3] The same sockets as [1] if required for that type of module to do dynamic changes (wether or not that change is then persisted to ini i don't know)

Changing the global configuration from GND->AIR should not be handled on a per-module basis i believe. I am really referring to setting shutter speed, chaninging metering mode etc. Not which port the camera is on, that is al 'static' config in ini files.

Beegad42, 2019-09-07:
What is actually broken with the SHM implementation (for sharing status info)? My understanding is that right now any process can access and update that status information via opening SHM via the file system.

webbbn, 2019-09-07:
IMO, the problem with it is that is requires configuring the shared memory page(s) before starting any of the programs, and it you can't see what the values are unless you connect to the shared memory. It also seems like a kluge, and there could be synchronization issues. You really should lock the shared memory before you read or write.

If we really think the efficiency of shared memory is required (IMO it's not), we could use ZeroMQ, which can use shared memory for local interfaces, but provides a standard, safe, easy to use interface.

That said, I don't know what's wrong with syslog, which has been used for decades.

raphaelscholle, 2022-10-16:
Since we now only use one executable, we do not need for that

closed 2022-10-16 by raphaelscholle