OpenHD/OpenHD #137: Languages to use

RespawnDespair, 2019-09-04:
As stated in #133 we use Python, Bash and C/C++ in the current core implementation.
I think it would benefit the project to choose only one language and i think we can actually port everything that is done to any of the languages. Speed wise my vote goes to C++.

So can we have a small tally in the replies for:

[1] Python and C/C++ 
[2] C/C++
[3] Let's not limit programming languages at all
[4] Other (specifiy the alternative)

RespawnDespair, 2019-09-04:
[2]

webbbn, 2019-09-04:
It seems unnecessary to limit the use of a few shell script where it makes sense, for simple startup scripts and such.

You're not going to get around using C/C++.

Forcing everything to be written in C/C++ would make some tasks ten times harder and make them harder to maintain and harder to debug.

My vote is for [3]. Use the best language for the task, which is python for control and communications and C++ for any processing intensive / time sensitive tasks.

RespawnDespair, 2019-09-05:
> My vote is for [3]. Use the best language for the task, which is python for control and communications and C++ for any processing intensive / time sensitive tasks.

That actually sounds a lot like [1]. Which on further investigation from me is also seeming more likely.
Pure C/C++ convolutes a lot of otherwise simple things. So i'll change my answer to [1]. I would really like to limit the languages used in order to prevent an uncontrollable mess :)

pilotnbr1, 2019-09-05:
To me it sounds like remove bash wherever we can with most appropriate language. But all things being equal, prefer python than c/c+

RespawnDespair, 2019-09-05:
Mixing SVPCOM/Brians ideas boils down to using Python instead of Bash for all governing and C/C++ for the core programs. Which is option [1]. Agreed?

webbbn, 2019-09-05:
Yes. I believe that is correct.

RespawnDespair, 2019-09-06:
Looks like [1] will be the winner here, i'll leave the issue open for a bit to allow more people to chime in.

Beegad42, 2019-09-07:
Option 1 is good.

RespawnDespair, 2019-09-07:
Well folks, seems this issue has been decided.
[1] It is, which incidentally is what svpcom uses, and several of us have expressed our love for that solution from a technical standpoint.

closed 2019-09-07 by RespawnDespair