OpenHD/OpenHD #138: Transmitting AirPi Desktop Screen to Ground Pi instead of camera feed

Brent1798, 2019-09-04:
I just want to start off and say this is an incredible project and the DIY RC community has benefitted greatly from it.

On the note of my issue, my question is simple. How would you transmit video of your desktop on the raspberry pi to the ground pi instead of streaming the camera footage directly?

I saw another issue very similar where someone asked how to transmit the camera footage after they had edited it with OpenCV Python. I tried reading that one, but I got a little lost in all the back and forth comments. If someone could provide a clear and conclusive answer for how to transmit a video stream from a python program through the Open.HD system instead of transmitting the camera video directly, I would greatly appreciate it.

Dealing with the original question if I know how to transmit a video I can simply screenshot my desktop and transmit that at an acceptable frame rate. Please help!

Brent1798, 2019-09-04:
Is there a way to transmit a saved video file using the Open.HD system?

webbbn, 2019-09-04:
The main program is called tx_rawsock, and it just reads raw h264 from a named pipe. If you stay around a standard 1280x720 frame size, you should be able to feed any raw h264 video into that pipe.

Brent1798, 2019-09-04:
Thank you so much @webbbn any idea where I should start in my quest to accomplish this?

webbbn, 2019-09-04:
It's a bit of a mess, but I think you should find it in this script: https://github.com/HD-Fpv/Open.HD/blob/master/wifibroadcast-scripts/tx_functions.sh

Brent1798, 2019-09-04:
I think I found the lines I'm looking for...
`echo "nice -n -9 raspivid -w $WIDTH -h $HEIGHT -fps $FPS -b $BITRATE -g $KEYFRAMERATE -t 0 $EXTRAPARAMS -o - | nice -n -9 /home/pi/wifibroadcast-base/tx_rawsock -p 0 -b $VIDEO_BLOCKS -r $VIDEO_FECS -f $VIDEO_BLOCKLENGTH -t $VIDEO_FRAMETYPE -d $VIDEO_WIFI_BITRATE -y 0 $NICS" >> /dev/shm/startReadCameraTransfer.sh
echo "nice -n -9 raspivid -w $WIDTH -h $HEIGHT -fps $FPS -b $BITRATE_10 -g $KEYFRAMERATE -t 0 $EXTRAPARAMS -o - | nice -n -9 /home/pi/wifibroadcast-base/tx_rawsock -p 0 -b $VIDEO_BLOCKS -r $VIDEO_FECS -f $VIDEO_BLOCKLENGTH -t $VIDEO_FRAMETYPE -d $VIDEO_WIFI_BITRATE -y 0 $NICS" >> /dev/shm/startReadCameraTransfer_10.sh
echo "nice -n -9 raspivid -w $WIDTH -h $HEIGHT -fps $FPS -b $BITRATE_5 -g $KEYFRAMERATE -t 0 $EXTRAPARAMS -o - | nice -n -9 /home/pi/wifibroadcast-base/tx_rawsock -p 0 -b $VIDEO_BLOCKS -r $VIDEO_FECS -f $VIDEO_BLOCKLENGTH -t $VIDEO_FRAMETYPE -d $VIDEO_WIFI_BITRATE -y 0 $NICS" >> /dev/shm/startReadCameraTransfer_5.sh
`
Which line is used depends on the bitrate, its line 312 - 314 in tx_functions.sh

Can anyone help me breakdown those lines? I feel like it's saying something like "take video from raspivid with these parameters, pipe into tx_rawsock with these parameters."If that is the case, how would I manipulate it to take a saved video file?

@webbbn @anyone

webbbn, 2019-09-04:
Just off the top of my head, you would replace the raspvid portion with something like:

ffmpeg -i <video file>-re -

The -re forces ffmpeg to stream the file in real time, not all at once. You might need some other parameters for specifying the output format (-vcodec copy -f h264).

Brent1798, 2019-09-04:
So after a little digging, I discovered "nice" at the beginning is used to set the value of importance in CPU time. By setting it to -9, you are setting it to relatively high priority. -n just tells you the next value is the value of CPU priority. I am thinking if I replace the section before "|" with some commands to play a video, that could work. 

Now if I wanted to go the route of openCV editing the video, then submitting it to the tx_functions.sh it would get a lot more complicated. Maybe a command instead of the raspivid section, it just grabs an image from a file location and sends that instead of a video stream. That way I could just have python and open cv edit the video stream, save the file to a location, and at the same time the Open.HD program is grabbing the last file saved and send that. There would probably be a significant lag due to saving times, but it is somewhere to start. 

@webbbn

Brent1798, 2019-09-04:
Okay so for this project it would be the most ideal to simply be able to screenshot my desktop and send that image over the system, what commands could I possibly use so that instead of taking the video from raspivid, it just constantly screenshots the desktop and sends that...

I'll keep researching

@webbbn

htcohio, 2019-09-05:
I actually have some interest in this as well.  Eventually it would be nice to insert various video sources and test h.264 files or at least have basic instructions/guidelines for doing so. 

P.S. there are inexpensive HDMI-CSI camera adapters one of which I recently got and sent my windows desktop screen over openhd.  The Main advantage is you have the simplicity and reliability of the MIPI csi-2 camera port.

Here is the Video.

https://youtu.be/SM8dHXmP0Nw

Brent1798, 2019-09-05:
Haha, I had actually considered rigging together two pis and simply sending the hdmi out from the main on the the CSI in of the transmitter, so I’m glad to know it worked well for someone. @htcohio 

I found a command line tool called “Scrot” which allows a user to screenshot the desktop or any window from the command line, which means we could basically use that instead of the raspivid section to do this, but it doesn’t help our problem of sending presaved videos, or sending videos from a python file

Brent1798, 2019-09-07:
Has anyone else thought of a way to transmit saved videos or a different video from like a python program?

RespawnDespair, 2020-12-04:
The original issue seems to have been addressed. Closing the issue.
If you feel the issue is not resolved, please reopen.

closed 2020-12-04 by RespawnDespair