OpenHD/OpenHD #139: AU8814 mavlink only receive, no uplink

GlidLov, 2019-09-04:
hi all,
using open.hd since early build towards 25/8 image, all them have same bug with mavlink telemetry and 8814 cards (alfa network 1900): telemetry can be read but not written. Anything which requires an uplink does not work: (i'm using QGroundControl and pixhawk cube 2.1 + arducopter 3.6.9), some examples:
- parameters don't load at all
- mission editor: if you add waypoints then click "upload" it tries then goes to an error, impossible to upload
- any info which requires to upload some data does not work (i.e. distance to home - which zeroes when armed, zeroing it requires uplink).

tried the very same setup, only swapped network card to atheros, and freq from 5.8 band to 2.4, everything works fine (up & down link)

RespawnDespair, 2019-09-04:
I don't have an 8814 to test, but can you confirm it is using the aircrack driver and not some standard driver in the kernel? So it should use the 88xx driver. I believe one way of achieving this is:

`ls /sys/class/net/wlan0/device/driver/module/drivers`

Where `wlan0` should be changed to the 8814.

htcohio, 2019-09-04:
Luke, I think this could have to do with Ziprays Frametype=2 commit where realtek uses RTS Frames for Video. 

Perhaps although that improves the downlink stability, it breaks the Uplink stability?

htcohio, 2019-09-04:
@GlidLove,

I am fairly certain that the RTL8814AU card should work on the ground with the same driver for 8812au. 

Also, can you confirm that you have identical images on both SD cards? If you use an older image with a new image, there's an encryption key that will not match that could cause the problem you're having.

Also, Can you confirm what you have for uplink stuff?
For these???

RANGEorENCRYPTION=
CTS=

If none of the previous solutions are the case or work you might try rolling back this one commit locally over SSH.  I am curious to find out if it improves on the problem or not. 
https://github.com/HD-Fpv/Open.HD/commit/1696cc848e974aa4ed4201c7baa08bd2377efe82

I hope your solution is somewhere in there.

GlidLov, 2019-09-05:
i have the same image (25/8) on both sides, everything is default but the frequency, which i changed as i wrote, FC baudrate to 57600 and debug=Y. i attach the file here. 
[openhd-settings-1.txt](https://github.com/HD-Fpv/Open.HD/files/3579952/openhd-settings-1.txt)
Copied the same to air unit.
Today i tried with a pair of brand new asus ac56, set to 5805mhz, they work fine both down & uplink.

steveatinfincia, 2020-12-07:
This is resolved for a while now, in OpenHD 2.0 there is a quirk setting since I didn't want to make major changes to anything. Set `FORCE_REALTEK_TELEMETRY_DATA_FRAME=Y` in the settings file and it works normally.

OpenHD 2.1 doesn't require any of this stuff and will work by default.

closed 2020-12-07 by steveatinfincia