OpenHD/OpenHD #143: Modular core system

RespawnDespair, 2019-09-07:
Heavily related to #136 and #134.

I have spent the last week trying to create some proof of concepts of a truly modular system. Which allows you to just drop a Python class in a folder and you can instantly use it in the open.hd system. But upon reflection that might be a bit overkill...

So here is my suggestions/brainfarts for the modular design of Open.HD:

- All specific logic is abstracted behind a generic interface
- NOWHERE in the core code should there be references to specific hardware
- If there is and edge case that requires an exception for a specific piece of hardware it should still be implemented in the generic interface allowing each specific module implementation to either ignore it or implement it in it's own way. It is also possible to throw an exception to let he system know this is not possible for that piece of hardware (e.g. very specific wifi transmission settings)
- A specific module can be SBC specific (e.g. a specific module for the Raspberry Pi camera that will only work on a Raspberry Pi SBC)
- When compiling a full image we need to be able to specifiy the SBC target to eliminate specific modules from the build

Current modules i can see are:

- Camera modules
- Stream modules (wrappers for the TX/RX programs with different settings for different kinds of streams, e.g. Telemetry and Video use different stream settings)
- Wifi cards (specific implementation of the setup and tear-down of the supported wifi cards)
- Plugins (Generic modules that can be used to start programs on GND or AIR)
  - I think OSD and Video playback could be implemented as generic modules

Modules are dependent on input and output. There should be an easy notation to pipe the output of a camera to the input of a stream. Likewise the output of the GND stream should be easily directed to the generic plugin for video playback or mavlink routing etc.
Possible ways to achieve this is using UDP streams or Named Pipes.

Modules can have specific configs for that module and are tied together in the main config, which basically sets up the different modules to be started and the 'streams' between them.

Please leave your thoughts and improvements.

RespawnDespair, 2019-09-07:
In addition to the above, there are special cases for the Wifi cards. The system currently decides which card to use for transmission on the ground bases on the card with the best RSSI. This problem is stated in #136. There should be a system in place that:

- Publishes a library that can be used in Python and C/C++
- Allows for the quick and easy reading of the relevant metrics of the wifi cards being used (rssi, settings, chip, etc)

In #134 we discuss the system that should be used for IPC, which will be the backing system for such a library.

So the Wifi module would have an extra set of requirements.

pilotnbr1, 2019-09-08:
Sounds really good!

closed 2023-01-31 by raphaelscholle