OpenHD/OpenHD #145: Issues when booting GroundPi with ethernet connected

Beegad42, 2019-09-11:
Noted on 09 Sept 2019 image but may not be a new issue (some symptoms seen in the past).
GroundPi is 3b+
Settings.txt is from image with just frequency changed to 5180 and UseMCS=Y UseSTBC=Y UseLDPC=Y

As is normal OSD Shows LAN IP obtained but may then show a message "OSD can not be created - check OSDConfig.txt for errors (may not be exact wording) - Then there is no OSD, just video

Other times OSD may be created OK but two copies of OSD are started resulting in some characters being blurry as the screen flicks between each OSD image output.

This is an intermittent issue that may be related to the spped at which an IP address is obtained via DHCP.

htcohio, 2019-09-11:
Thanks for the info.

Does Ethernet Tethering or whatever you were using it for still work?

Beegad42, 2019-09-11:
If the OSD shows the Ethernet IP during boot then yes it works, it is all a bit flakey as if I plug the ethernet in after boot it doesn't work.

careyer, 2019-09-11:
Same same problem here.... I compiled my image 2019-08-29 ....
I get exactly the same behaviour and error messages as @Beegad42 reports

I think it is safe to classify this a bug.

Beegad42, 2019-10-09:
Confirm,ed that this is a timing bug, as per comments above. The issue only occurs if  the Pi does get an ethernet IP via DHCP,. If the Pi fails to get an IP (which takes longer) the this issue does not occur.

pilotnbr1, 2019-10-10:
It may be noting that there is a difference in “sleep” timing between Openhd main.sh and EZ-wbc when it comes to the Ethernet stuff. It’s like a 3 second difference in sleep.

Why are we even shutting off the osd for any of this? Why don’t we just eliminate the shutdown calls and the restart calls

RespawnDespair, 2020-12-04:
These issues should be long fixed by now. Closing issue.
If these problems still exist, feel free to reopen the issue.

closed 2020-12-04 by RespawnDespair