OpenHD/OpenHD #156: AeroSim USB dongle with Flysky FS-iA6B Receiver - FlySky FS-i6 Transmitter

VU2ATX, 2019-09-14:
**//AeroSim USB dongle with Flysky FS-iA6B Receiver. On your FlySky FS-i6 Transmitter, assign Ch5 to the 3 position.
//switch on the upper right side of the transmitter. And assign Ch6 to the 2 position switch located on upper
//right, towards the edge. In the mission planner, you will have the flight mode CH5 corresponding to the 3 position
// switch and CH7 corresponding to the 2 position switch on your Flysky FS-i6. 
//DO NOT CHANGE THE FOLLOWING, IT IS WELL TESTED WITH "Open.HD_V2,0_RC1,1-2019-09-09.img"** 

#define ROLL_AXIS      0
#define PITCH_AXIS     1
#define YAW_AXIS       5
#define THROTTLE_AXIS  6
#define AUX1_AXIS      2
#define AUX2_AXIS      7
#define AUX3_AXIS      3
#define AUX4_AXIS      4

VU2ATX, 2019-09-14:
So this way, you do not need an expancive Frsky 9XD transmitter and also you will get rid of physical USB connection.

You will have a 3 position switch to change three flight modes (say: Stablize, Auto, RTL) and 2 position switch which is mapped here on Ch7 so that in mission planner, you could assign any additional function to it like; ALT HOLD or LAND or Gripper etc. or could use it for camera switching.

Above setting and be included in the joyconfig.txt file.
@Dev. team: pls consider it and use it in fourthcomming work.

VU2ATX, 2019-09-14:
[image]

careyer, 2019-09-15:
This is a neat setup! However it has the drawback that you will considerably interfere with a 2,4Ghz Open.HD setup due to sending 2,4Ghz FlySky RC to the receiver. It likely works in close proximity but on longer flights you will likely see problems here.
This is the beauty of the direct USB-connection to the 9XD. Here the FrSky transmitter gets completely deactived and cannot interfere with the system ;-)

VU2ATX, 2019-09-15:
You are right and I expected that it will be said. I also have X9D and I know its bulky and expensive, though it does not interfere due to RF shut down. In the setup I shown, one can select frequencies quite away using AR9271 based wifi adapters so that interference is minimal.

Since its wireless, transmitter can be kept away from Ground Pi and Transmitter antenna can be wrapped in alluminium foil to reduce rf power.

closed 2019-09-16 by VU2ATX