OpenHD/OpenHD #157: RC control with multi-card setup

ABLomas, 2019-09-15:
TL;DR - i think that current uplink antenna switching logic is not correct - open.hd does not pick correct antenna (with higher signal strength) for RC signal uplink.

I tried to make debug.txt by setting DEBUG=Y in config but no debug.txt appeared on SD card after boot and 5mins of waiting, so cannot attach it. If you tell me which script generates this info - i can connect keyboard and just run it manually.

So, about a problem - i used ali blue sticks and DIY clovers - 1x on air, 2x on ground. Range (video + RC control via ibus) was approx 4km.
Swapped one antenna to helical, video uplink shows one card with ~10dBm stronger signal  as expected, but RC link drops together with clover signal.
Tested 5 ali blue sticks, all had similar problems. OK, maybe they are bad.

Swapped setup to 3x AWUS036NHA cards - 1x in air, 2x on ground.
Testing for range:
[image]
Looks ok (for omnidirectional antennas), had link and video loss soon after but i think 5km for omnidirectional antennas is not bad.
Returned, swapped one clover on ground with helical. Took off:
[image]
So, directional antenna has perfect signal (and no dropouts on video), omni antenna has not so good signal and RC uplink has very bad signal with dropouts even at 3.7km range.
Regained signal and tried to fly again:
[image]
As you can see i get no video signal on omni antenna (maybe possible as it was partially shadowed by helical) and RC signal strength is weaker than video downlink on helical.
Fighting with failsafes i reached my max with this combo:
[image]
(then i lost RC link and RTH'ed home, RC control was not regained until ~1.5km from home)
As you can see in last screenshot - i had video downlink on both cards, video was good, without glitches, but RC control uplink was lost.

So, with directional + omni antenna combo i got slightly lower RC range compared to both omni setup, but perfect video.
I think, in current implementation without explicit "PrimaryCardMAC=" configuration open.hd selects wrong card to send RC uplink signals.
It should pick card with highest signal strength (best antenna).

Another user on Telegram having similar problems:
> Oscar, [14.09.19 16:31]
> So using two differents antennas (omni and directional) is uneless if using RC with Openhd? I have failsafes while perfect vídeo with directional antenna...

Also, this is easy to replicate on ground using attenuator chain and 50ohm terminators - using attenuator + antenna and terminator on ground and terminator on plane - RC signal is lost before video is lost (even in the same room).
And yes, ofc there is possibility that i had 4 bad ali blue sticks in a row and out of the box broken AWUS036NHA...

ABLomas, 2019-09-15:
I want to add, that even on ground tests i observe the same thing - most of time uplink signal strength is between weaker and stronger signal (but sometimes drops below both).
This is consistent with idea that wrong card is selected for sending (lower signal strength, but smaller uplink packets still can reach destination) OR switching works correctly but something is wrong with uplink power.

AWUS036NHA + 50ohm terminator as air unit, AWUS036NHA + 50ohm terminator at 50cm distance as 1st ground unit, AWUS036NHA + 26 + 18dB attenuators + clover as third unit working as "directional antenna":
[image]
This time i captured signal strength below both units, but most of time it is in middle (-85dBm if numbers taken from screenshot). If "stronger card" is sending packets - it should be above -84dBm, right?

htcohio, 2019-09-16:
@ABLomas,

Thank you very much for the thorough information.  I want you to know that I'm investigating it within the next day or two and I have not forgot about you :) 

P.S. about debug I did not mean to get it during flight as a matter of fact you probably want to have debug disabled during a flight because it uses CPU.

I just wanted to see if you could attach a copy of your settings text file at least if the debug text file was not on the SD card so I could look at the settings. 

If possible, just attach your latest settings here and, I have RF attenuators as well as the same cards as you so it will be very easy to set up a test environment.

Cheers

ABLomas, 2019-09-16:
Ground and air settings file are identical, so i attach it here:
[openhd-settings-1.txt](https://github.com/HD-Fpv/Open.HD/files/3617805/openhd-settings-1.txt)

Thanks

htcohio, 2019-09-17:
Hello,

I was unable to recreate your results.  From what I was able to see, the automatic switching works perfectly. 

I was able to recreate your results using a blown Blue stick that had really good RSSI reception but was not able to help put any power.

So, when the RC on the ground goes by the lowest receive our RSSI the card with the best signal on the ground could be the one with the blown amplifier or limited power output for some reason. 

You mentioned that you used Alfa Cards after discovering that one of your blue cards could be bad, so I would imagine it should work well for you. 

But, Over the last month there have been a couple changes on things for other card frame-types.  I I'm going to share a link to the latest image (9/13) that had a couple corrections. 

If you have a chance would you be able to check it out and do a quick comparison?

ABLomas, 2019-09-17:
I used latest at this time (few days old) image and screenshots done when flying with AWUS036NHA cards.
Also the same replicated on completely different setup (another RPI set, another set of cards, another plane).
Sure i can try another image if you provide link, but so far i think RC over open.hd is usable with single antenna, not diversity - but i re-test.

htcohio, 2019-09-17:
Cool,

I would be happy to take a video of my test setup.  It just takes some
time.

FYI I confirmed the auto switching at the same Gatorade and exact settings
you had.

Also, I forgot that I already saved new img to my drive the other day, you
just need to set the same settings that you had on it.

Here is the link. (9/13 img)
https://drive.google.com/file/d/1At0VqIk9M9JYXNNLgGcrGBZdekAU5HZb/view?usp=drivesdk

Justin Healey
President
HTC Communications LLC
O 440-523-1400
C 440-523-9578
www.htcohio.com

ABLomas, 2019-10-22:
OK, i re-did everything with "proper" setup ;-)
Based on telegram chat - most of my issues are caused by rpi0 and blue sticks. So i took and re-did everything from scratch. In field we had two planes:
1) One with new setup:

1. Ground side - RPI 3B+
2. Air side - RPI 3A+
3. brand new camera cable, properly fixed and even glued in place. Additional power from ground cards, filtered SBEC on air/ground, etc.
4. Ultra-low resolution (below 360p) to reduce bandwith and CPU load (config attached - still using full sensor readout, just scaled down; probably lowest possible widescreen resolution with simple (fast) scaling according to forums)
5. DIY cloverleaf air side, DIY clover + helical ground side (all tuned to be < 1.3 VSWR at used frequency, 17km tested on analog video without glitches)
6. AWUS036NHA cards (1 in air, 2x on ground)
7. Control via Open.HD RC (Taranis via USB)

2) One plane with old setup:

1. Ground side - RPI 3B+
2. Air side - RPI Zero
3. Using 720p resolution, default image setup except OSD part
4. DIY clover air, DIY clover + helical ground, tuned; Furious CP patch
5. Ali blue sticks (3x on ground, 1x in air)
6. Control via external LRS

I already verified, that i can have 5km range using omni (cloverleaf) antennas without diversity. Assuming noise and other factors - 2km using clovers (no diversity) is easy task, without packet loss, etc. This was tested several times using both setups.
Now let's add diversity into play (first setup, described above):
https://youtu.be/IjbGU5B2xF0

- at 0:57 i suddenly lost control - video was without glitches, RSSI good enough, range barely 1.1km (almost 5x lower that just by using single omni antenna setup). RTH initiated return home (OSD incorrectly shows flight mode from INAV, it's RTH, not STAB). Kinda strange to loose signal at -70dBm, no?
 - while still flying back - uplink RSSI still goes into "- -" sometimes, even when video is fine. Again, strange - using single antenna setup i did not noticed such issues.
 - at 1:32 - another "very-big-latency" event (~2 seconds latency), but this is for another discussion
 - at 2:43 i lost control again, this time 150m from me (i still had omni antenna attached) and video few seconds later.
 - another retry for range, at 5:57 i lost signal again, ~1.8km range, video still without issues.

Testing the same setup with single cloverleaf (or dual, identically oriented clovers) do not show such RC/video signal loss issues, i got clear 2km (did not tested more this time, maybe could get the same 5km).
Test with "old" plane (setup #2) shows approx the same pattern just with video-only:
 - using single-antenna - 2km easy (video)
 - using multiple directional + omni antennas - barely 500m (video)

In my opinion - diversity (multiversity) switching for antennas is completely broken. Maybe using single directional antenna on tracker is viable solution for long range + RC, but using multiple antennas just kills range (video/rc) and that's all.

In also captured telemetry after flight (the same in youtube video). rcpackets.png looks strange, no?
All files from telemetry + config for setup #1 attached in this comment.

[openhd_rc_control_issues.zip](https://github.com/HD-Fpv/Open.HD/files/3758205/openhd_rc_control_issues.zip)

HD-Fpv, 2019-10-23:
Hi,

Just a few questions/Notes.

1. There have been some discussions saying that RC being used with CTS=N actually works better but this has not been thoroughly tested in the same way as your valuable test flights are.  Perhaps you would be able to confirm or deny this considering that your Fail-Safe seems to be working really well.

2. Just to confirm, you had Pi3A+ on air correct?

3. My personal use has always been with Mavlink RC + pixhawk.  Your flight is the first comprehensive review with inav so we can't rule out that being problematic. 

4. I believe Datarate 3 + CTS=Y might be a better combination because rate 3 (12mbps) actually uses the RTS frames 

So, if you are able to find time....
-- Try again with the exact same settings however, use Datarate 3 instead...

-- Next, try Datarate 3 with CTS=N

Also, Rodizio once said that setting the Thresh62 lower to 24 could help improve RC.

You can change this SSH and accessing ath9k_fw paramaters.

ABLomas, 2019-10-23:
Hi,
about #2 - yes, i had pi3A+ on air side.
I think it works in the same way as arduplane - some time earlier i hooked IBUS output to pixhawk controller and see the same "rx loss" in logs (just log analyzer does not show this in pretty way).
I believe that "rcpackets.png" and FC logs are identical - packet lost - FC registers RX loss (and the same on pixhawk), but this is long way until failsafe (rc signal should be lost for 2 seconds or so).
Even on ground, 10m TX<-->RX i see lost packets on FC side, example from INAV logs - yellow line shows "rxSignalReceived" value:
[image]
So, i think, it's not FC<-->openhd interoperability issue.

About more flights - i think, i can do on weekend, but would be very good (and usefull when flying) if @pilotnbr1 could add "FAILSAFE" flag as alert. Or as icon, or any other way - currently OSD do not show flight mode change on failsafe (probably on INAV only), so only way to notice it - when plane starts flying "on it's own".
Symbol, alert or anything on screen would help to identify failsafe condition earlier.

Beegad42, 2019-10-24:
You appear to have at least 4 issues.
1. Loss of RC when Groundpi switches from Helical to Clover/Omni.
2. Inconsistent Telemetry arriving at ground - looks like may be in Air (Inav?)
3. High packet loss which appears to get worse as time passes (i.e. even as you get closer to home).
4. Video latency.

Many of which can be caused by interference which actually makes your test flights valuable as it is a real world test of the code.

htcohio, 2019-10-24:
Hi,

Could uou look at this recent zipray commit? (He also has a PR for RSSI).

Basically, if CTS=Y then RTS frames should be used for RC. 

Does this still appear to be the case?

You may remember that there was some debate related to the opposite scenario wheras (with 8812) RC using Standard Data frames caused less packet loss on the bench... 

I just want to see if anything changed with the frame type since this commit.  

My concern is that just because packet loss appears to be less on the bench with RC using standard data frames doesn't necessarily mean it's best for long-range flights. 
https://github.com/HD-Fpv/Open.HD/commit/1696cc848e974aa4ed4201c7baa08bd2377efe82

htcohio, 2019-10-24:
> You appear to have at least 4 issues.
>
> 1. Loss of RC when Groundpi switches from Helical to Clover/Omni.
> 2. Inconsistent Telemetry arriving at ground - looks like may be in Air (Inav?)
> 3. High [...]

Yes, real world scenario is very valuable indeed thank you very much we will get to the bottom of this for sure Andreas!

ABLomas, 2019-10-24:
So, for next test case i will test, so far (to limits):

1. single card on ground, omni antennas on both ends
2. dual-card, omni+directional antenna setup, rc fixed in setup (MAC) to omni antenna
3. dual-card, omni+omni setup without fixed cards in config (range should be the same as in #1 in theory but i doubt so)

Anything else to add? Justin, i know how to ssh or just connect keyboard/screen and change settings. Just don't know if this would be consistent with other testings (in other words, something probably should be changed, but maybe after all other tests/before and re-do everything?

BG: interference possible, for sure. In my flying place i have strong (actually very strong considering that there are no big cities around - probably standing between two pencil beams on datalink) spike on 2340MHz, some ppl also flying around ~30m away from me with frsky (2400-2480 FHSS). But i do not expect to get sudden 10x increase in latency just because interference. Also, all those interference sources listed above won't disappear anywhere - i rarely fly alone, so there will be other users around (not so close - even analog FPV @2370MHz do not "feel" those transmitters) and i can't control external datalinks.

Beegad42, 2019-10-24:
I am not sure if you have the hardware but it would be interesting to see how much packet loss you get at 5.8GHZ with external RC . If that is good then start adding back things like RC etc.

ABLomas, 2019-10-25:
I have 4x  AC56 and other hardware, but again, this requires complete rework.
I can also add external RC (433M LRS), but this already kills whole idea and easier here just go with good old analog video - 20km+ range with almost zero delay, no unexpected latency, perfect WDR on cams, lighter and smaller setup, etc. Just OSD not so impressive ;-)
So, i would like to finish with this setup, if it proves to be unusable in real world scenario (for example, external interference causes unexpected delays over 1 second, etc), then i will just ditch whole plane and go with another system ;-)

htcohio, 2019-10-25:
Hey, I understand the frustration just understand a lot of compatibility
and changes have been made over the last few months so, were just trying to
tweak the settings to get them right.

With pi zero and Russian wifibroadcast image, you can try to use that
without the bells and whistles to see how it should normally work.

RespawnDespair, 2020-12-04:
Since some of the discussion in this thread appears to still be relevant in the current 2.0 branch I have attached this issue to the stability effort for 2.0

We need to check the relevant parts of this discussion and take a good look at the code. I have had several inquiries about the CTS=Y/N setting for RC and it's relevance. If nothing else we need to clarify why things are the way they are.

closed 2023-01-31 by raphaelscholle