OpenHD/OpenHD #158: Camera switching issue in latest OpenHD

VU2ATX, 2019-09-16:
1、
RPi Air:
RPi <-> RPi camera
RPi <->  Ethernet IP camera
RPi <-> USB 8812AU 
RPi Ground:The toggle switch is in third gear.Android app rtp port 5601.
RPi <-> USB 8812AU
RPi <-> Android phone+HDMI monitor
Everything is all right
2、
RPi Air:
RPi <-> RPi camera
RPi <-> Ethernet IP camera
RPi <-> USB 8812AU 
RPi Ground:The toggle switch is in third gear.Android app rtp port 5601.
RPi <-> USB 8812AU
RPi <-> Android phone（without HDMI monitor）
I can't switch to the  IP camera (only) mode.The switch on  third position.
but I can switch to the  IP camera on  Android phone mode.The switch is on  second place.
3、
RPi Air:
RPi <-> RPi camera
RPi <-> USB camera
RPi <-> USB 8812AU 
RPi Ground:The toggle switch is in third gear.Android app rtp port 5601.
RPi <-> USB 8812AU
RPi <-> Android phone（without HDMI monitor）
Everything is all right

_Originally posted by @zipray in https://github.com/HD-Fpv/Open.HD/issues/114#issuecomment-511258006_

Hi, I am following the wiki To test camera switch (RPI native camera + virtual USB\IP).

image

My Setup:
RPi Air:
RPi: Raspberry Pi 2B
CSI2 Camera: PiCam 5mp, V1
USB Camera: Logitech C920
WiFi USB Adapter: AR9271L based adapter with external antenna.

GroundPi:
RPi: Raspberry Pi 3B+
WiFi USB Adapter: AR9271L based adapter with external antenna.
Display:
1. Android Phone connected via USB teathering for Video / Telemetry QGroundControl apk
installed.
2. HDMI cable also ready to be connected to the HDMI monitor.
Joystick: Transis X9D connected via USB Cable. Ch7 is ready with PWM min=1000 and PWM max=1999 to be used to switch camera.

I am not getting to what you said : Toggle switch in third gear, there is nothing about it on wiki, I am refering to.

I am not getting success in camera switch however I am getting video when using single USB camera.
Waiting to hear from you.

VU2ATX, 2019-09-16:
[image]

htcohio, 2019-09-17:
Hey,

One thing, is it a Pi2b or 2b+ air? 

I think the Pi zero and Pi2b have the same single core cpu and different kernel config. 

That might be a limiting factor at the moment. 

I will check, but plz confirm if its a pi2b or 2b+?

VU2ATX, 2019-09-17:
> Hey,
> 
> One thing, is it a Pi2b or 2b+ air?
> 
> I think the Pi zero and Pi2b have the same single core cpu and different kernel config.
> 
> That might be a limiting factor at the moment.
> 
> I will check, but plz confirm if its a pi2b or 2b+?

Hi, I have just physicall open cases of both Pi's and observed that My AirPi (Pi 3B v1.2) and My GroundPi (Pi 3B+)
[image]
[image]

VU2ATX, 2019-09-17:
First image is of AirPi and second image is of GroundPi

htcohio, 2019-09-17:
Ok, thanks.

That set up should work.

Like I said, you might want to experiment with the resolution settings.

Or, you can try this older image that Vitaliy shared after he wrote the
code.  Maybe something broke it since then.  Here is the link to yhe 7/30
version that should have been the first one.

https://drive.google.com/file/d/1Md5VI73eWxMiad8-5jDPmEWXjQiU_1ww/view?usp=drivesdk

VU2ATX, 2019-09-17:
Okay thanks, will give it a try and update

VU2ATX, 2019-09-18:
Hi, **the bottomline is, I got 80% success. getting both videos (from Picam and from C920)** in regular QGroundControl installed from Google play store, QGC_V3.4_17.06_h264_or_RTP.apk is not working, I reported earlier also, pls see the error message I reported.

Storry:
I have tried a lot the image "Open.HD-2019-07-30_GROUND_ONLY.img"  and did not find anything special which could resolve my issue then I went back to "Open.HD_V2,0_RC1,1-2019-09-09.img".

I started doing various iterations and during one of the iterations, I set ChannelIPCamera=5 which is a 3 position switch on my transmitter, IsBandSwicherEnabled=1, ChannelToListen2=7 is a 2 position switch on my transmitter, ChannelToListen=8 (there is no ch8 in my transmitter).

When I was changing the flight mode with 3 position switch on my transmitter which is ch5, I observed, the round blue leds of the USB camera C920 started glowing (this was ultimate joy for me).

I changed in android the udp port in QGCS to 5601 and video from C920 came perfect then I changed the udp port back to 5600 and video came from Picam.

However by changing 2 position switch CH7, nothing happened. My next aim is to view both video streams using QGC_V3.4_17.06_h264_or_RTP.apk which has not worked for me.

VU2ATX, 2019-09-18:
Hi, I have fiddled the openhd-settings-1.txt bit more and finally able to switch USB camera with a 2 position switch. Though QGC_V3.4_17.06_h264_or_RTP.apk still did not work for me. My setting is as follows;

**# My setup
# Logitech C920 HD Wrbcam and PiCam 5MP V1 camera on AirPi
# Two AR9271L based wifi adapters (one each side; GroundPi, AirPi) with external antenna.
# Pi Models: GroundPi=Pi3B+, AirPi=Pi3B
# Video monitor: Android Phone with QGCS**

FREQ=2312
DATARATE=2

VIDEO_BLOCKS=8
VIDEO_FECS=4
VIDEO_BLOCKLENGTH=1024

WIDTH=1280
HEIGHT=720
FPS=30

KEYFRAMERATE=5

UseMCS=N
UseSTBC=N
UseLDPC=N

Bandwidth=10

FC_RC_SERIALPORT=/dev/serial0
FC_RC_BAUDRATE=57600
FC_TELEMETRY_SERIALPORT=/dev/serial0
FC_TELEMETRY_BAUDRATE=57600

TxPowerAir=27
TxPowerGround=27 
# For long range, you can assign power 58 

IsBandSwicherEnabled=1
**ChannelToListen2=8 
# Ch8 is not available on my transmitter Flysky FS-i6**

SecondaryCamera=USB
**ChannelIPCamera=7
# Ch7 is a two position switch on my transmitter Flysky FS-i6**. Actually FS-i6 is a 6 channel Transmitter and I have re-mapped Ch6 to # Ch7 in the joyconfig.txt to make this 2 position switch (Ch7) useful for autopilot (Pixhawk). 

**# Modified USB camera pipeline for video resolution 800x600**

USBCamera="gst-launch-1.0 uvch264src device=/dev/video0 initial-bitrate=\$1 average-bitrate=\$1 iframe-period=1000 name=src auto-start=true src.vidsrc ! queue ! video/x-h264,width=800,height=600,framerate=30/1 ! h264parse ! rtph264pay ! udpsink host=127.0.0.1 port=5600"

IsCamera1Enabled=1
IsCamera2Enabled=1
IsCamera3Enabled=0
IsCamera4Enabled=0

**ChannelToListen=8
# Ch8 is not available on my transmitter Flysky FS-i6**

Camera1ValueMin=995
Camera1ValueMax=1005
Camera2ValueMin=1995
Camera2ValueMax=2000

htcohio, 2019-09-18:
> Camera1ValueMin=995 look

Hello

Ahh hey some success!

Ok, so it seems to be working mostly but you don't have a lot of Axis
channels for the 3 functions.

First, here is a newer 9/13 test image, if you have not tried this one yet,
would you be able to try it and also set a joystick or knob so the 3
position is tested?

As follows:

IsCamera1Enabled=1
IsCamera2Enabled=1
IsCamera3Enabled=1
IsCamera4Enabled=0
**ChannelToListen=4 (or any axis...

Camera1ValueMin=995
Camera1ValueMax=1350
Camera2ValueMin=1355
Camera2ValueMax=1650
Camera3ValueMin=1656
Camera3ValueMax=2000

Please try out the functions of the QGC App with this image.

There was a previous image version that had problems with the UDP hotspot
splitter not sending the video.

https://drive.google.com/file/d/1At0VqIk9M9JYXNNLgGcrGBZdekAU5HZb/view?usp=drivesdk

VU2ATX, 2019-09-18:
Hi, I was able to view both videos one by one with udp port setting to 5600 and then 5601 in standard QGCS from Google Play store and I had connected the android phone to the GroundPi via wifi hotspot and was finding no issue like "UDP hotspot splitter not sending the video".

The only problems I am facing at the moment:
QGC_V3.4_17.06_h264_or_RTP.apk not working in my phone and giving error message of local library which I thing has nothing to do with your latest image of 13th Sept.

There is one more issue observed today that when I opt to use sdcard for video recording then no video saved on the USB pen drive however if I select memory option then video is saved but only of PiCam and not of USB cam.

At the moment, I am downloading the 13th Sept image and will test tomorrow. I have Transis X9D a standard recomended transmitter to act like a joystick, will use that, it has many channels, can assign a pot (knob) to required channel.

What are the various other enhancements in the 13th sept image ?

htcohio, 2019-09-18:
Hi,

Mostly, nothing that you would notice directly.  Some of the Telemetry RSSI
transmission values are sent using MCS/STBC modulation with 8812au cards.

But, I also had an issue a few weeks ago where the custom qgc app was only
"white screen" without video.

The custom qgc app is a really nice feature with the multi-camera setup
like you have because it provides Picture-in-picture in the app.

VU2ATX, 2019-09-19:
hey, what you have done in 13th sept image ! Camera switching not happening with similar settings as that which were working for 9th sept image.

VU2ATX, 2019-09-19:
Hi, did desktop testing with the latest image "Open.HD_V2,0_RC1,1-2019-09-13.img"
Its too working perfect with my above setting. 

For the first time QGC_V3.4_17.06_h264_or_RTP.apk displayed video from USB camera C920 however it did not display the video from Picam.

htcohio, 2019-09-19:
Hey,

Sorry, I'm a little confused about what's working and what's not working,
your last two messages kind of contradict each other lol

htcohio, 2019-09-19:
Nevermind, it looks like you deleted a previous message...

About Pi cam, did you return the default 5600 port on QGC?

Also, if you are using default "rtp" instead of "raw" you might need to uncheck "UDP h264 inside QGC (see pic)

So, if RAW...Check box...

IF RTP...... Uncheck box...

 

[image]

VU2ATX, 2019-09-19:
Hi, yes, 13th Sept image is working well. For Picam, let me check and try what you suggested.

closed 2019-09-19 by VU2ATX